p2rkw/xf86-input-mtrack #1: Finger is only detected when I start the movement from the right.

botto, 2015-07-24:
If I try to move the mouse when I place my finger just a bit to the left instead of starting on the right side of the trackpad I get no movement. It will not register my finger until it has moved within the right side of the trackpad.

p2rkw, 2015-07-24:
I guess it's related to new EdgeSize property which I was added long ago on 'dev" branch which I merged into master recently.

Probably origin of your trackpad's coordinate system in its central point.
Can you make few experiments to confirm my guess?
You can get your actual EdgeSize property value with
```
xinput list-props 11 | grep Edge
```
(just replace 11 with your trackpad id).
Then you can set it to 0 so exactly half of your touchpad will be inactive.
Value of 50 should completely disable your touchpad.
You can change value of this property with following command:
```
xinput set-prop 11 "Trackpad Edge Size" 4
```
where last integer is new property value.
Can you try values of 0 and 50 and post your results?

p2rkw, 2015-07-24:
Can you test this commit: https://github.com/p2rkw/xf86-input-mtrack/commit/6f03cc5d504ef1db4fbe88b1a4277d4899ec5e25 ?

botto, 2015-07-24:
Yup, that worked.
Also changing the edgesize values worked correctly.

closed 2015-07-24 by p2rkw

p2rkw, 2017-09-02:
There is slight chance that commit https://github.com/p2rkw/xf86-input-mtrack/commit/549c66135ead1dcd6a68517f36ee9615a8c4c3dd will reintroduce this issue. Let me know if so.