p2rkw/xf86-input-mtrack #3: Kinetic scrolling

mck182, 2015-08-26:
It would be awesome if mtrack could do what synaptics/OS X do when scrolling - when the scroll gesture is fast and short, it keeps the view scrolling even after the fingers are lifted from the touchpad while putting the fingers back on the touchpad stops the scrolling.

Or is this already possible? I haven't figured out how to set that...

Btw. thanks for your new work on mtrack, good stuff!

p2rkw, 2015-08-27:
Something like that: https://www.youtube.com/watch?v=AdOUelRzDP8 ?
It doesn't look like something hard to implement, but I'm afraid it may be not as smooth as on OSX, because mtrack can only generate mouse button events.
How does it work with Synaptics driver? did you tested it?

mck182, 2015-08-27:
Yeah that's pretty much it.

It works rather nicely, smoothly even, with Synaptics. You can basically test on any live distro as those mostly ship Synaptics. I'm not sure how it works exactly though.

Apparently this is called "Coasting" in Synaptics, according to http://askubuntu.com/questions/338841/configure-50-synaptics-conf-for-kinetic-scrolling

I see a branch "coast" already exists, maybe Ryan already started it? I'll try that branch and see if that does anything useful.

mck182, 2015-08-27:
Ok so I've managed to build that branch and as far as I can say it doesn't work with "ScrollCoastEnable" "true", plus it makes X go crazy on CPU and the mouse pointer freezes altogether for few seconds.

But then again I'm not sure if that is what it is meant to be doing :)

p2rkw, 2015-09-15:
Ding dong! New features arrived!

Please test how coasting and smooth scrolling works for you.
Feature branch is here: https://github.com/p2rkw/xf86-input-mtrack/tree/coasing2

mck182, 2015-09-16:
Woah, nice!

Ok so from quick testing it works rather nicely with Qt/KDE apps, it's not as smooth in Chromium (though that's Chromium), but the "kinetic" element is there (ie. make the gesture fast and it moves a lot).

Though I think the scroll could go a liiiiitle bit longer perhaps.

I'll keep it running for some days and report back.

Thanks a lot!

p2rkw, 2015-09-23:
I pushed few new commits, added new configuration properties. 

I think is ready to push to master.

mck182, 2015-09-24:
Awesome, will try in a bit.

So far it seems pretty good. It still has some room for improvements but now with the config values, it might just be perfect.

Thanks a lot!

closed 2015-09-26 by p2rkw