p2rkw/xf86-input-mtrack #4: ScrollSmooth does not work together with natural scrolling

terlar, 2015-10-27:
I have configured natural scrolling by switching the binding for `ScrollUpButton` and `ScrollDownButton`.

e.g.
```
    Option "ScrollUpButton" "5" # 2 finger swipe up to scroll down
    Option "ScrollDownButton" "4" # 2 finger swipe down to scroll up
```

I find that when you have `ScrollSmooth` enabled this doesn't work and I assume it also interfears with the smooth scrolling causing a lot of jerkiness, since they are counterworking eachother.

p2rkw, 2015-10-27:
Can you test it, please?

terlar, 2015-10-29:
Yes, this seems to be working now. Thank you for the quick fix. I have now enabled smooth scrolling and will come back if I find any other weird issue.

closed 2015-10-29 by terlar