p2rkw/xf86-input-mtrack #6: Touchpad continues to scroll when movement is not happening

gsora, 2015-11-09:
When I 2-finger scroll, the scrolling continues even if I stopped the movement itself.

How can I disable this behaviour?

p2rkw, 2015-11-09:
I don't know what might cause it.
I need to see your configuration file and xinput dump for touchpad device.

gsora, 2015-11-09:
Here's the xinput dump:

```
gsora@djentleman ~ % xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ bcm5974                                 	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Apple Inc. Apple Internal Keyboard / Trackpad	id=10	[slave  keyboard (3)]
```

[Here's](https://gist.github.com/gsora/32d6de025481698c9ea8) my configuration.

p2rkw, 2015-11-09:
I thought about dump for your touchpad device. Can you please pasete output of
```xinput --list-props 11``` ?

gsora, 2015-11-09:
Here you go:

```
gsora@djentleman ~ % xinput --list-props 11
Device 'bcm5974':
	Device Enabled (136):	1
	Coordinate Transformation Matrix (138):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (272):	0
	Device Accel Constant Deceleration (273):	1.000000
	Device Accel Adaptive Deceleration (274):	1.000000
	Device Accel Velocity Scaling (275):	10.000000
	Trackpad Disable Input (276):	0
	Trackpad Sensitivity (277):	0.400000
	Trackpad Touch Pressure (278):	5, 5
	Trackpad Button Settings (279):	1, 1
	Trackpad Button Emulation Settings (280):	0, 1, 100
	Trackpad Button Emulation Values (281):	3, 2, 0
	Trackpad Tap Settings (282):	50, 120, 400
	Trackpad Tap Button Emulation (283):	1, 3, 2, 0
	Trackpad Thumb Detection (284):	1, 0
	Trackpad Thumb Size (285):	25, 70
	Trackpad Palm Detection (286):	1, 1
	Trackpad Palm Size (287):	40
	Trackpad Gesture Settings (288):	10, 100
	Trackpad Smooth Scroll (289):	1
	Trackpad Scroll Settings (290):	150, 20, 0
	Trackpad Scroll Buttons (291):	5, 4, 6, 7
	Trackpad Swipe Settings (292):	700, 300, 0
	Trackpad Swipe Buttons (293):	8, 9, 0, 0
	Trackpad Swipe4 Settings (294):	700, 300, 0
	Trackpad Swipe4 Buttons (295):	0, 0, 9, 8
	Trackpad Scroll Coasting (296):	0.100000, 180.000000
	Trackpad Scale Distance (297):	150
	Trackpad Scale Buttons (298):	12, 13
	Trackpad Rotate Distance (299):	150
	Trackpad Rotate Buttons (300):	14, 15
	Trackpad Hold1Move1 Stationary Settings (301):	20, 1
	Trackpad Hold1Move1 Settings (302):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (303):	1, 1, 1, 1
	Trackpad Drag Settings (304):	0, 350, 40, 200
	Trackpad Axis Inversion (305):	0, 0
	Trackpad Edge Size (306):	10
```

p2rkw, 2015-11-09:
First guess: something wrong with touch pressure, can you try these settings:
```xinput set-prop 11 "Trackpad Touch Pressure" 5 6```
and 
```xinput set-prop 11 "Trackpad Touch Pressure" 6 5```
?

Does this reproduce with Smooth scroll disabled like this:
``````xinput set-prop 11 "Trackpad Smooth Scroll" 0``` 
?

I never observed such behaviour... I'm afraid I will need some "remote debugging".

gsora, 2015-11-10:
With smooth scrolling disabled this behaviour cannot be reproduced, so there's definitely something going on there.

Anything else I can report?

terlar, 2015-11-10:
I also experience this, not using the smooth scrolling now again. Could also help out debugging.

p2rkw, 2015-11-10:
Good to hear it's reproducible.

This issue may be related to just coasting. To check that you can enable smooth scroll but disable coasting:
```
xinput set-prop 11 "Trackpad Smooth Scroll" 1
xinput set-prop 11 "Trackpad Scroll Coasting" 2.0 0.0
```

First argument of "Trackpad Scroll Coasting" property is minimal scroll speed required to trigger coasting, second is it's duration. Setting duration to 0 will disable coasting. If it will help, you can try to re-enable it and check how first value changes behavior of scrolling.

terlar, 2015-11-10:
If I do:

```
xinput set-prop 11 "Trackpad Scroll Coasting" 2.0 0.0
```

Then scrolling stops working completely.

gsora, 2015-11-10:
I'll report as soon as I get to my computer. 

@terlar for sake of completeness, can you post the output of 

    xinput --list-props 11

gsora, 2015-11-10:
I can confirm that with 

    xinput set-prop 11 "Trackpad Scroll Coasting" 2.0 0.0

scrolling doesn't work anymore.

p2rkw, 2015-11-10:
Ok, but with smooth scroll disabled or not?

gsora, 2015-11-10:
Either with smooth scrolling enabled or disabled.

japostoles, 2015-11-29:
I'm seeing this issue as well

p2rkw, 2015-11-30:
I'm working on this.

Ruin0x11, 2016-02-02:
I'm still having this problem. Are there any updates?

p2rkw, 2016-02-02:
I reproduced, and identified this problem, but I have to refactor existing implementation a bit. I hope I will push something by end of this week.

p2rkw, 2016-02-07:
Re-factorization in progress. If it is really annoying for you try disable smooth scrolling as described above.

p2rkw, 2016-02-08:
Fix on branch [github-6](https://github.com/p2rkw/xf86-input-mtrack/tree/github-6)

gsora, 2016-02-08:
Great, thank you, I'll test these changes tomorrow and I'll let you know

p2rkw, 2016-02-11:
Did somebody tried it? Does it work? I need your results to push it to master.

Ruin0x11, 2016-02-11:
It looks like it works for me. Thanks for the fix.

commit dd3f7315 (2016-02-13): Introduce completly new timers system.

Fix few issues with coasting. By the way fixed #6.
closed 2016-02-13 by p2rkw