p2rkw/xf86-input-mtrack #8: Not working

LordTaifleh, 2015-11-27:
Well, i have to say, i'm quite a beginner, so beware.

tl;dr touchpad not recognized, no logs available(!?)

i cloned this git to my arch machine and did
./configure
sudo make && make install
which gave me no errors. Two files are output to /usr/local/lib/xorg/modules/input
(i also tried copying those to another directory to make X find them. yet, nothing)
Then i configured the already existing 90-libinput.conf in /usr/share/X11/xorg.conf.d according to the Readme of this git (Identifier Touchpads, driver mtrack, matchistouchpad on)

after logoff or restart, the cursor wont move. i tried to look for errors in the log, but there's no log in /var/log and journalctl _COMM=Xorg doesn't give anything either. Also $xinput list does not list the device at all. Oh Xorg -version is 1.18.

Am I doin something not quite right?

p2rkw, 2015-11-27:
To compile and install you should run `make && sudo make install`. How did you install driver without root privileges?
Usually log is written to /var/log/Xorg.0.log. For faster diagnose you can attach it here.
Alternatively you can try AUR: https://aur.archlinux.org/packages/xf86-input-mtrack-git/

There is also a chance that mtrack doesn't work with Xorg 1.18: https://github.com/p2rkw/xf86-input-mtrack/issues/7

Please make sure you have correctly installed mtrack, and paste Xorg logs here.

LordTaifleh, 2015-11-27:
Well, i did install with root, seems i missplaced the sudo in the text. but i did do this to install, since, the other way round, it just gave me errors.
The second part about the log is where shit gets weird. there is no log. whereis cant find  it and it surely isnt in /var/log. 

thanks, i will try again with the AUR version.
I did not, at first, because i thought that it was an other branch, the one by bluedragon, and i am not aware of the differences in the capabilities in the drivers.

p2rkw, 2015-11-27:
Arch wiki says logs may be placed in `~/.local/share/xorg/`, can you please check that location?

Once you try AUR package please let me know how it works with 1.18.

LordTaifleh, 2015-11-27:
I confirm that, for me, AUR mtrack works on Xorg 1.18. it behaves weird though, but isnt yet configured.
Also, the log appeared, don't ask why it didn't in the first place.

LordTaifleh, 2015-11-27:
Not trying to be an utter pain, but ...
The cursor is jumping around when moving, like being snapped to a grid. Setting mouse acceleration to zero makes it better, but then the cursor just moves real slow. I've tried to find the optimal settings, but to no extent. the cursor always seem really "nervous".
Are the GUI settings interfering?
can i set acceleration or general speed in 10-mtrack.conf also?

closed 2015-11-27 by LordTaifleh