p2rkw/xf86-input-mtrack #10: cannot seem to use left/right physical button

ngoctnq, 2016-01-06:
As the title said - I cannot find the settings for the physical button, and multifinger gestures are a hit-and-miss with my sweaty palms. Is there anything of which I should know?

p2rkw, 2016-02-08:
As always: please post output of 'xinput --list-props' for your touchpad device.

ngoctnq, 2016-02-09:
Hi, sorry for sounding too abrupt in the first message. The output your command gave is:

```
ngoc ~ $ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ bcm5974                                 	id=14	[slave  pointer  (2)]
⎜   ↳ HID 05ac:820b                           	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ HID 05ac:820a                           	id=11	[slave  keyboard (3)]
    ↳ Apple Inc. Apple Internal Keyboard / Trackpad	id=13	[slave  keyboard (3)]
ngoc ~ $ xinput --list-props 13
Device 'Apple Inc. Apple Internal Keyboard / Trackpad':
	Device Enabled (150):	1
	Coordinate Transformation Matrix (152):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Product ID (269):	1452, 610
	Device Node (270):	"/dev/input/event9"
```

p2rkw, 2016-02-10:
Try 'xinput --list-props 14'. Your 'internal keyboard' wasn't picked by mtrack.

ngoctnq, 2016-02-10:
I'm using the rMBP 15" Mid-2012, if it offers any bonus insight.
Here is the output:

```
ngoc ~ $ xinput --list-props 14
Device 'bcm5974':
	Device Enabled (150):	1
	Coordinate Transformation Matrix (152):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (273):	0
	Device Accel Constant Deceleration (274):	1.000000
	Device Accel Adaptive Deceleration (275):	1.000000
	Device Accel Velocity Scaling (276):	10.000000
	Trackpad Disable Input (305):	0
	Trackpad Sensitivity (306):	0.650000
	Trackpad Touch Pressure (307):	10, 1
	Trackpad Button Settings (308):	1, 1
	Trackpad Button Emulation Settings (309):	0, 0, 100
	Trackpad Button Emulation Values (310):	1, 3, 2
	Trackpad Tap Settings (311):	50, 120, 400
	Trackpad Tap Button Emulation (312):	1, 3, 2, 0
	Trackpad Thumb Detection (313):	1, 0
	Trackpad Thumb Size (314):	25, 70
	Trackpad Palm Detection (315):	1, 0
	Trackpad Palm Size (316):	40
	Trackpad Gesture Settings (317):	10, 100
	Trackpad Smooth Scroll (318):	1
	Trackpad Scroll Settings (319):	150, 20, 0
	Trackpad Scroll Buttons (320):	5, 4, 7, 6
	Trackpad Swipe Settings (321):	1, 0, 1000
	Trackpad Swipe Buttons (322):	1, 1, 1, 1
	Trackpad Swipe4 Settings (323):	700, 300, 0
	Trackpad Swipe4 Buttons (324):	0, 0, 0, 0
	Trackpad Scroll Coasting (325):	0.100000, 180.000000
	Trackpad Scale Distance (326):	150
	Trackpad Scale Buttons (327):	12, 13
	Trackpad Rotate Distance (328):	150
	Trackpad Rotate Buttons (329):	14, 15
	Trackpad Hold1Move1 Stationary Settings (330):	20, 1
	Trackpad Hold1Move1 Settings (331):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (332):	1, 1, 1, 1
	Trackpad Drag Settings (333):	0, 350, 40, 200
	Trackpad Axis Inversion (334):	0, 0
	Trackpad Edge Size (335):	10
```

p2rkw, 2016-02-13:
Try: xinput set-prop 14 "Trackpad Button Emulation Settings" 1 0 100

Then search for 'ButtonZonesEnable' in README.md

closed 2016-02-13 by p2rkw

ngoctnq, 2016-05-28:
Sorry for way too late of a reply, but back then I completely removed Linux because I got annoyed and hasty-- after reinstalling, and following the steps listed above, it works now. Thank you!