p2rkw/xf86-input-mtrack #11: Crashes kdeinit5

psi-4ward, 2016-01-20:
Hi there,

I'm on Arch and installed the https://aur.archlinux.org/packages/xf86-input-mtrack-git/ Pkg.
```
Linux psi-nb 4.3.3-2-ARCH #1 SMP PREEMPT Wed Dec 23 20:09:18 CET 2015 x86_64 GNU/Linux
kwin 5.5.3
plasmashell 5.5.3
```
and SDDM.

After installing the AUR Package i can not login into Plasma anymore.

Xorg.0.log
```
[     8.287] (**) Option "Device" "/dev/input/mouse1"
[     8.287] (EE) mtrack: cannot configure device
[     8.287] (EE) Couldn't init device "SynPS/2 Synaptics TouchPad"
[     8.287] (II) UnloadModule: "mtrack"
```

I couldn't find anything helpful in journald log.

p2rkw, 2016-05-03:
Do you have rights to read/write to /dev/input/mouse1? Are you member of input group?
You can add some logs and compile it by yourself, this will guarantee that your compilation is compatible with your X installation.

From this short description, I don't know what's going on, but you are not the only one having this issue. I think you could ask on arch forums.

p2rkw, 2016-10-25:
Is it still reproducible?

psi-4ward, 2016-10-26:
No! Seems to work now. Thanks!

I assume `/dev/input/mouse1` was from the connected external mouse.

closed 2016-10-26 by psi-4ward