p2rkw/xf86-input-mtrack #12: Whenever I change settings using `10-mtrack.conf` or `xinput set-prop ...` one finger Tapping become weird

dontdieych, 2016-01-31:
weird means, it looks like button down event triggered. But without additional move or touch, it just holding. So double tap or tap-and-slight-move needed for actual tapping.

It happens whenever I change custom option in `/etc/X11/xorg.conf.d/10-mtrack.conf` or using `xinput set-prop`.

Thanks.

dontdieych, 2016-01-31:
> aura -Q xf86-input-mtrack-git xorg-server
xf86-input-mtrack-git 0.4.0.r2.g04c222e-1
xorg-server 1.18.0-4

p2rkw, 2016-01-31:
What exactly did you changed?

dontdieych, 2016-01-31:
> xinput set-prop 12 "Trackpad Smooth Scroll" 0

funny thing is ... it's ok right now :).

I'll tell if I get same problem again.

dontdieych, 2016-01-31:
It happens again. Restart X then OK now.

How could I tell this reasonably...

p2rkw, 2016-02-10:
Which button is hold? From 3 to 6?
Can you check it with 'xinput --test xxx'? Run it, then focus another terminal window using only keyboard and run set-prop there.

dontdieych, 2016-02-11:
https://drive.google.com/file/d/0B27A327bOmbYSDk0OGx6YmEtQUE/view?usp=sharing

`xcalc` button black means holding down. It should be white after tap. But It looks like,

- first touch -> press (holding)
- second touch -> release, press (holding)
- and so on

For couple fo days, I was doing bisect manually by hand.

Almost done. I'll tell you when I find that commit.

p2rkw, 2016-02-13:
Can you test is it still reproducible on current master branch?

dontdieych, 2016-02-14:
I've used master branch around 20 hours. So far so good.

Thanks!

closed 2016-10-12 by p2rkw