p2rkw/xf86-input-mtrack #13: Thumb detection

markolenik, 2016-02-09:
First of all: Thank you for forking mtrack. You have fixed so many issues for me!

Maybe I'm just using wrong settings but whenever I touch the bottom edge of the trackpad with my thumb _after_ my index finger is on the pad, the motion is stopped for a short time. Is there any way to completely ignore the bottom edge?

Below my configuration (most of it is default).

    Section "InputClass"
      Identifier "Touchpads"
      Driver "mtrack"
      MatchIsTouchpad "on"
      MatchDevicePath "/dev/input/event*"
      Option "Sensitivity" "0.38"
      Option "TapButton1" "1"
      Option "TapButton2" "3"
      Option "TapButton3" "2"
      Option "TapButton4" "0"
      Option "ClickFinger1" "1"
      Option "ClickFinger2" "3"
      Option "ButtonMoveEmulate" "false"
      Option "ButtonIntegrated" "true"
      Option "ButtonEnabled" "true"
      Option "ClickTime" "25"
      Option "BottomEdge" "50"
      Option "FingerHigh" "3"
      Option "FingerLow" "1"
      Option "IgnoreThumb" "true"
      Option "IgnorePalm" "true"
      Option "TapDragEnable" "false"
      Option "TapDragWait" "100"
      Option "ScrollUpButton" "4"
      Option "ScrollDownButton" "5"
      Option "ScrollDistance" "1"
      Option "SwipeDistance" "75"
      Option "SwipeLeftButton" "1"
      Option "SwipeRightButton" "1"
      Option "SwipeUpButton" "1"
      Option "SwipeDownButton" "1"
      Option "SwipeClickTime" "0"
      Option "SwipeSensitivity" "1000"
      Option "ThumbSize" "20"
    EndSection

p2rkw, 2016-02-10:
It's "EdgeSize" now, instead of "BottomEdge"

markolenik, 2016-02-10:
Great, thank you!

closed 2016-02-11 by p2rkw