p2rkw/xf86-input-mtrack #15: Unwanted tap clicks still occur after setting "TapButton" settings to 0

Ruin0x11, 2016-02-25:
I'm using Arch Linux on a 2012 MacBook Air with the xf86-input-mtrack-git package. Despite setting all `TapButton` settings to 0, I'm still getting extraneous tap-clicks when I leave one finger on the touchpad and touch a second finger to it. This issue can be reproduced in `xinput test` with one finger on the touchpad and touching a second finger to it repeatedly, causing several spurious sequences of `button press 1` immediately followed by `button release 1` with several `motion` events between them.

I might have accidentally missed something in the config, but trying to tap with a single finger or two fingers at once causes no button presses.

Here is the output of `xinput list-props`:
```
Device 'bcm5974':
	Device Enabled (138):	1
	Coordinate Transformation Matrix (140):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (271):	0
	Device Accel Constant Deceleration (272):	1.000000
	Device Accel Adaptive Deceleration (273):	1.000000
	Device Accel Velocity Scaling (274):	10.000000
	Trackpad Disable Input (275):	0
	Trackpad Sensitivity (276):	0.500000
	Trackpad Touch Pressure (277):	5, 5
	Trackpad Button Settings (278):	1, 1
	Trackpad Button Emulation Settings (279):	0, 1, 100
	Trackpad Button Emulation Values (280):	3, 2, 0
	Trackpad Tap Settings (281):	50, 120, 1
	Trackpad Tap Button Emulation (282):	0, 0, 0, 0
	Trackpad Thumb Detection (283):	0, 0
	Trackpad Thumb Size (284):	25, 70
	Trackpad Palm Detection (285):	0, 0
	Trackpad Palm Size (286):	40
	Trackpad Gesture Settings (287):	10, 100
	Trackpad Smooth Scroll (288):	1
	Trackpad Scroll Settings (289):	250, 20, 0
	Trackpad Scroll Buttons (290):	5, 4, 7, 6
	Trackpad Swipe Settings (291):	700, 300, 0
	Trackpad Swipe Buttons (292):	8, 9, 10, 11
	Trackpad Swipe4 Settings (293):	700, 300, 0
	Trackpad Swipe4 Buttons (294):	0, 0, 0, 0
	Trackpad Scroll Coasting (295):	0.100000, 200.000000
	Trackpad Scale Distance (296):	150
	Trackpad Scale Buttons (297):	12, 13
	Trackpad Rotate Distance (298):	150
	Trackpad Rotate Buttons (299):	14, 15
	Trackpad Hold1Move1 Stationary Settings (300):	1, 0
	Trackpad Hold1Move1 Settings (301):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (302):	1, 1, 1, 1
	Trackpad Drag Settings (303):	0, 1, 100, 200
	Trackpad Axis Inversion (304):	0, 0
	Trackpad Edge Size (305):	10
```

And `10-mtrack.conf`:
```
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier      "Touchpads"
    Driver          "mtrack"
    Option          "Sensitivity" "0.50"
    Option          "TapButton1" "0"  
    Option          "TapButton2" "0"
    Option          "TapButton3" "0"
    Option          "TapButton4" "0"
    Option          "MaxTapMove" "0"
    Option          "ScrollUpButton" "5"
    Option          "ScrollDownButton" "4"
    Option          "ScrollLeftButton" "7"
    Option          "ScrollRightButton" "6"
    Option          "ScrollDistance" "250"
    Option          "TapDragEnable" "no"
    Option          "TapDragTime" "0"
    Option          "TapDragWait" "100"
    Option          "Hold1Move1StationaryButton" "0"
    Option          "Hold1Move1StationaryMaxMove" "1"
EndSection
```

Ruin0x11, 2016-03-04:
I set `TapDragDist` to 0 also, but nothing changed. I've already set `TapDragEnable` to `no` and set `Hold1Move1StationaryButton` to 0, so both should be nullified.

Ruin0x11, 2016-05-05:
I'm still having this issue, having accidentally clicked on things multiple times.

p2rkw, 2016-05-11:
I will take a look at this once I find some free time.

paulthomson, 2016-10-04:
I think I can also reproduce this:

1. Rest one finger on the trackpad.
2. Repeatedly tap a second finger on the trackpad while keeping the original finger on the trackpad. You will end up "clicking" even though this should not occur.

Ruin0x11, 2016-10-04:
Yes, I'm still having this issue also.

p2rkw, 2016-10-12:
Probable "Hold1Move1" feature mess here a bit. I think I'll add another variable to make turning it off easier/possible.

p2rkw, 2016-10-13:
I had to rewrite tapping_update function from scratch because it was very hard to modify.
You can test results on task branch here: https://github.com/p2rkw/xf86-input-mtrack/tree/github-15

paulthomson, 2016-10-13:
Firstly, I am sorry. I did not have "Hold1Move1StationaryButton" "0". But I can now confirm that this did not work for me and now it does! Thank you so much! 

Issue solved for me!

paulthomson, 2016-10-13:
Perhaps I could also suggest that Hold1Move1StationaryButton is 0 by default, as it is very easy to trigger without understanding what is happening.

Ruin0x11, 2016-10-13:
Confirmed working for me also. Thank you for solving the issue.

p2rkw, 2016-10-13:
As for experimental features like hold&move: I think I have to make them enabled by default, otherwise I will not get any feedback on them, because nobody will ever notice. I feature will become annoying for someone there is  always explained in readme how to disable/tune features. If you have any thoughts on them create new issue for it.

I still have few new ideas, but I still not sure should I detach/rename or just bump version number, because some of them may break configuration file.

Fix is on master now.

closed 2016-10-13 by p2rkw

BlueDragonX, 2016-10-17:
@p2rkw any interest in submitting PR's back upstream?