p2rkw/xf86-input-mtrack #17: scrolling horizontallly while scrolling vertically (and vice-verse)

swalladge, 2016-04-23:
As the title says - and same as [https://github.com/BlueDragonX/xf86-input-mtrack/issues/108](https://github.com/BlueDragonX/xf86-input-mtrack/issues/108)

Only seems to be a problem when swapping the button numbers for scrolling to achieve natural scrolling. - Scrolling works as expected with the following:

```
    Option          "ScrollUpButton" "4"
    Option          "ScrollDownButton" "5"
    Option          "ScrollLeftButton" "6"
    Option          "ScrollRightButton" "7"
```

I originally commented there, but discovered that I was running this fork. Interestingly, I don't get the issue with BlueDragonX's git version, only with this version.

p2rkw, 2016-05-03:
Fix pushed, reopen if it won't work for you.

closed 2016-05-03 by p2rkw

swalladge, 2016-05-04:
Unfortunately I can't seem to get natural scrolling to work now (by swapping the scrollup/down buttons and scroll left/right).

p2rkw, 2016-05-04:
Hmm, I will look at it later, meanwhile you can try to reconfigure buttons on the fly with xinput. It may help.

p2rkw, 2016-05-04:
https://github.com/p2rkw/xf86-input-mtrack/commit/02fcf6e30290b0dadfbf55ac266745886c489051 should fix it.

swalladge, 2016-05-05:
Yep, works perfectly now, thanks. :D

commit f70ab768 (2016-05-03): Proper initialization and setup of scroll axes(2 and 3).

Till now axel 2 and 3 weren't initialized, add 'initAxle' to init them.
Swap axle 2 and 3, so 2 is now vertical scroll, 3 - horizontal.
Move hack to detect osx like scrolling from trigger_swipe_unsafe to updateSwipeValuators.
It's way less 'hackish' now.

Hope this will fix #17 on github.