p2rkw/xf86-input-mtrack #18: Moving cursor + thumb-click triggers 2-finger click

EiNSTeiN-, 2016-04-24:
I have a Surface Book and just configured mtrack and noticed this gesture does not work as well as someone might be used to on a OS X device.

When I move the cursor with a finger and thumb-click (holding the click down, such as when highlighting text) the right-click menu shows up just as if I had done a 2-finger click on the trackpad. This is unfortunate because it means I need to lift my finger completely off the trackpad before thumb-clicking whenever I want to highlight text.

It would be a nice improvement if fingers that are not in the bottom edge area were not considered part of the 2-finger click, so the finger moving the mouse does not have to come off the trackpad when clicking.

Here's my `xinput --list-props`

```
Device 'Microsoft Surface Keyboard Touchpad':
	Device Enabled (137):	1
	Coordinate Transformation Matrix (139):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (267):	0
	Device Accel Constant Deceleration (268):	1.000000
	Device Accel Adaptive Deceleration (269):	1.000000
	Device Accel Velocity Scaling (270):	10.000000
	Trackpad Disable Input (299):	0
	Trackpad Sensitivity (300):	0.650000
	Trackpad Touch Pressure (301):	12, 1
	Trackpad Button Settings (302):	1, 1
	Trackpad Button Emulation Settings (303):	0, 1, 100
	Trackpad Button Emulation Values (304):	3, 2, 0
	Trackpad Tap Settings (305):	50, 120, 400
	Trackpad Tap Button Emulation (306):	0, 0, 0, 0
	Trackpad Thumb Detection (307):	1, 0
	Trackpad Thumb Size (308):	25, 70
	Trackpad Palm Detection (309):	1, 0
	Trackpad Palm Size (310):	40
	Trackpad Gesture Settings (311):	10, 100
	Trackpad Smooth Scroll (312):	1
	Trackpad Scroll Settings (313):	40, 20, 0
	Trackpad Scroll Buttons (314):	4, 5, 6, 7
	Trackpad Swipe Settings (315):	700, 300, 0
	Trackpad Swipe Buttons (316):	0, 0, 8, 9
	Trackpad Swipe4 Settings (317):	700, 300, 0
	Trackpad Swipe4 Buttons (318):	0, 0, 0, 0
	Trackpad Scroll Coasting (319):	0.100000, 800.000000
	Trackpad Scale Distance (320):	150
	Trackpad Scale Buttons (321):	12, 13
	Trackpad Rotate Distance (322):	150
	Trackpad Rotate Buttons (323):	14, 15
	Trackpad Hold1Move1 Stationary Settings (324):	20, 1
	Trackpad Hold1Move1 Settings (325):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (326):	1, 1, 1, 1
	Trackpad Drag Settings (327):	1, 350, 40, 200
	Trackpad Axis Inversion (328):	0, 0
	Trackpad Edge Size (329):	18
```

EiNSTeiN-, 2016-04-26:
This happens a less often when I change the `ThumbRatio` to 25, but still happens quite frequently depending on the angle of my thumb on the trackpad. I still think it might be best to ignore all clicks that happen in the bottom edge area, I can't think of a reason they would be considered part of a gesture...

sxmichaels, 2016-05-20:
I think the Synaptics driver (and possibly libinput as well?) has a setting that defines a bottom region such that if a touch starts in that region it is ignored, but if another touch moves into that region it is still counted. That seems like what Mac is doing too; perhaps something like that could be implemented?

PR #20 (2016-05-29): Do not consider edge clicks when calculating emulated buttons
Fix #18 

This prevents fingers that are located in the edge area from being considered part of a 2- or 3-finger click. This fixes an annoying behavior on touch pads with integrated buttons, where moving one finger near the center of the touch pad and clicking with the thumb near the bottom of the touch pad would be considered a 2-finger click.

I believe this behavior will make a lot more sense for users with large touch pads, like the ones on Apple laptops and Surface Book laptops. 

@p2rkw
closed 2016-05-29 by p2rkw