p2rkw/xf86-input-mtrack #19: Disable h-scroll during v-scroll and vice versa

cytopia, 2016-05-11:
Is it possible to disable horizontal scrolling while you scroll vertically and vice versa?
This seems to be the standard behavior on OSX with two-finger scroll and is kind of easier.

It happens a lot to me, if I am using half a browser width and pages have a big horizontal scroll, that once I scroll up/down I also move left/right by accident.

Any thoughts on this?

p2rkw, 2016-05-12:
Are you sure you using current master?
It was fixed in https://github.com/p2rkw/xf86-input-mtrack/issues/17

cytopia, 2016-05-12:
I definitely have the above mentioned commit (when I check via `git log`) in your repo I compiled from source .

Anything else I can check? Any debug option I can turn on?

p2rkw, 2016-05-12:
In this case I will have to first reproduce and then debug it. Please post your configuration (as described in readme)

rmanne, 2017-01-24:
I still have this issue, was this resolved?

Output of `xinput --list-props 12` (bcm5974):
[xinputprops.txt](https://github.com/p2rkw/xf86-input-mtrack/files/725549/xinputprops.txt)

My configuration (not minimal) `10-mtrack.conf`:
[10-mtrack.txt](https://github.com/p2rkw/xf86-input-mtrack/files/725551/10-mtrack.txt)

I'm on the latest commit (eb3b9c3...).

p2rkw, 2017-01-27:
Is h-scroll triggered only at the bottom of the scrolled window? Which button events are generated (xinput --test <devid>)?

rmanne, 2017-01-27:
No, when v-scrolling, h-scroll triggers whereever, even in the center of a webbrowser page . When scrolling up for example, it's some button press/release 5 intersperced with some button press/release 6. Partial output here: [log.txt](https://github.com/p2rkw/xf86-input-mtrack/files/735457/log.txt)

p2rkw, 2017-01-28:
Could you please also attach output of `xinput list --long 12`. I'm still struggling with reproduction.
Does this horizontal movement have same speed(distance) as vertical, or is it rather occasionally?

rmanne, 2017-01-28:
To be completely honest, I thought it was initially it was the "default" effect. It's usually smooth and perfectly consistent. It jitters a lot when I move two fingers in a circle, but if it's just slight movements, mostly vertical, it's perfectly smooth. I made a gif to show the interaction: 
[image]

Here's the output of `xinput list --long 12`:
[log.txt](https://github.com/p2rkw/xf86-input-mtrack/files/737207/log.txt)

p2rkw, 2017-01-29:
So, this is how SmoothScroll(aka. pixel perfect scroll) works: what we observe is just "inaccurate" v-scroll.
Looks like I will have to add another configuration variable to ignore "perpendicular" movement while other is active.

p2rkw, 2017-01-29:
Please check this commit: https://github.com/p2rkw/xf86-input-mtrack/commit/fe087109c55e02c982a2d9c3ef229c8efd1726dc

rmanne, 2017-01-30:
Solid, it works.

p2rkw, 2017-01-30:
Slightly modified fix landed on master.

closed 2017-01-30 by p2rkw