p2rkw/xf86-input-mtrack #21: Cannot configure device

ArchBin4ry, 2016-06-25:
I can't seem to get mtrack driver loaded, using a HP Laptop Synaptics TouchPad that support 4 finger click + 3 finger swipe in windows

`[binary@hermes ~]$ sudo cat /var/log/Xorg.0.log | grep "(EE)"
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     3.981] (EE) BUG: triggered 'if (dev == ((void *)0))'
[     3.981] (EE) BUG: exevents.c:2131 in SetScrollValuator()
[     3.981] (EE) 
[     3.981] (EE) Backtrace:
[     3.982] (EE) 0: /usr/lib/xorg-server/Xorg (SetScrollValuator+0x1c4) [0x526834]
[     3.983] (EE) 1: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x7d97) [0x7f8746258db7]
[     3.983] (EE) 2: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x66da) [0x7f874625600a]
[     3.983] (EE) 3: /usr/lib/xorg-server/Xorg (xf86DeleteInput+0x1e2) [0x486672]
[     3.984] (EE) 4: /usr/lib/xorg-server/Xorg (config_fini+0x102b) [0x4993fb]
[     3.984] (EE) 5: /usr/lib/xorg-server/Xorg (config_fini+0x1613) [0x49a683]
[     3.985] (EE) 6: /usr/lib/xorg-server/Xorg (config_init+0x9) [0x497af9]
[     3.985] (EE) 7: /usr/lib/xorg-server/Xorg (InitInput+0xbf) [0x47adbf]
[     3.985] (EE) 8: /usr/lib/xorg-server/Xorg (remove_fs_handlers+0x421) [0x43a611]
[     3.988] (EE) 9: /usr/lib/libc.so.6 (__libc_start_main+0xf1) [0x7f874f597741]
[     3.989] (EE) 10: /usr/lib/xorg-server/Xorg (_start+0x29) [0x4245f9]
[     3.990] (EE) 11: ? (?+0x29) [0x29]
[     3.990] (EE) 
[     3.990] (EE) BUG: triggered 'if (dev == ((void *)0))'
[     3.990] (EE) BUG: exevents.c:2131 in SetScrollValuator()
[     3.990] (EE) 
[     3.990] (EE) Backtrace:
[     3.990] (EE) 0: /usr/lib/xorg-server/Xorg (SetScrollValuator+0x1c4) [0x526834]
[     3.991] (EE) 1: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x66da) [0x7f874625600a]
[     3.991] (EE) 2: /usr/lib/xorg-server/Xorg (xf86DeleteInput+0x1e2) [0x486672]
[     3.991] (EE) 3: /usr/lib/xorg-server/Xorg (config_fini+0x102b) [0x4993fb]
[     3.992] (EE) 4: /usr/lib/xorg-server/Xorg (config_fini+0x1613) [0x49a683]
[     3.992] (EE) 5: /usr/lib/xorg-server/Xorg (config_init+0x9) [0x497af9]
[     3.992] (EE) 6: /usr/lib/xorg-server/Xorg (InitInput+0xbf) [0x47adbf]
[     3.993] (EE) 7: /usr/lib/xorg-server/Xorg (remove_fs_handlers+0x421) [0x43a611]
[     3.994] (EE) 8: /usr/lib/libc.so.6 (__libc_start_main+0xf1) [0x7f874f597741]
[     3.994] (EE) 9: /usr/lib/xorg-server/Xorg (_start+0x29) [0x4245f9]
[     3.995] (EE) 10: ? (?+0x29) [0x29]
[     3.995] (EE) 
[     3.996] (EE) mtrack: cannot configure device
[     3.996] (EE) Couldn't init device "SynPS/2 Synaptics TouchPad"
[   591.881] (EE) kernel bug: Touch jump detected and discarded.
[   591.901] (EE) kernel bug: Touch jump detected and discarded.
[ 12273.977] (EE) kernel bug: Touch jump detected and discarded.
[ 12277.168] (EE) kernel bug: Touch jump detected and discarded.
[ 12277.168] (EE) kernel bug: Touch jump detected and discarded.
[ 12277.580] (EE) kernel bug: Touch jump detected and discarded.
[ 12277.580] (EE) kernel bug: Touch jump detected and discarded.

`

p2rkw, 2016-06-26:
Try this: https://github.com/p2rkw/xf86-input-mtrack/issues/11

p2rkw, 2016-06-26:
As a workaround please try set "ScrollSmooth" to 0.

ArchBin4ry, 2016-06-27:
It still shows the same error, may I know what can I provide for you to debug? My laptop model is HP Spectre X360 13-4103DX. I tried xev, only move, tap, click, scroll, pinch is recognized; 2 finger click, 3 finger swipe, 4 finger swipe all does not recognize.

p2rkw, 2016-06-27:
Error message is similar to those in issue #11 . Can you attach whole error log file? I guess mtrack for some reason tries to open your mouse device. Also please attach your minimal config file.

For future investigation you could compile mtrack in debug mode just to check real names of these two functions:
```
1: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x7d97) [0x7f8746258db7]
2: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x66da) [0x7f874625600a]
```

gammy, 2016-07-20:
Hm. I have a similar issue on an Apple Trackpad. This is unfortunately not on a debug build: [mtrack_cannot_configure_device.txt](https://gist.githubusercontent.com/gammy/8d561f6a135bd728470b575cc94f6720/raw/ca0863a597301f6abac39c8f3942a4ccc573743b)

gammy, 2016-08-02:
I have built the driver will full debugging `--enable-debug` but I see no additional output (I still just get a pointer reference regarding `SetScrollValuator`). Can this be a bug in the configure script, or is there little/no debugging during initialization? Debugging the driver during load (via `gdb` or similar) is beyond my expertise. This bug is a shame for it means I can of course no longer use the mtrack driver at all. Is anyone watching this thread?

p2rkw, 2016-10-11:
Fixed by https://github.com/p2rkw/xf86-input-mtrack/commit/c29e7fa879d514b7039375c960ffb0b6df3ed48b
(pls report if not)

closed 2016-10-11 by p2rkw

lewishsh, 2017-01-16:
I have a similar issue, with the latest version. Here's the unclipped xorg.log

[  1008.442] 
X.Org X Server 1.19.1
Release Date: 2017-01-11
[  1008.455] X Protocol Version 11, Revision 0
[  1008.459] Build Operating System: Linux 4.8.14-1-ARCH x86_64 
[  1008.463] Current Operating System: Linux lhmacbook 4.8.13-1-ARCH #1 SMP PREEMPT Fri Dec 9 07:24:34 CET 2016 x86_64
[  1008.463] Kernel command line: \vmlinuz-linux root=/dev/sda5 rw add_efi_memmap irqpoll initrd=/intel-ucode.img initrd=/initramfs-linux.img
[  1008.471] Build Date: 12 January 2017  07:11:40AM
[  1008.476]  
[  1008.480] Current version of pixman: 0.34.0
[  1008.488] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[  1008.488] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1008.504] (==) Log file: "/home/lewis/.local/share/xorg/Xorg.0.log", Time: Mon Jan 16 09:27:47 2017
[  1008.508] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  1008.508] (==) No Layout section.  Using the first Screen section.
[  1008.508] (==) No screen section available. Using defaults.
[  1008.508] (**) |-->Screen "Default Screen Section" (0)
[  1008.508] (**) |   |-->Monitor "<default monitor>"
[  1008.508] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[  1008.508] (==) Automatically adding devices
[  1008.508] (==) Automatically enabling devices
[  1008.508] (==) Automatically adding GPU devices
[  1008.508] (==) Max clients allowed: 256, resource mask: 0x1fffff
[  1008.508] (WW) The directory "/usr/share/fonts/Type1/" does not exist.
[  1008.508] 	Entry deleted from font path.
[  1008.508] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/100dpi/".
[  1008.508] 	Entry deleted from font path.
[  1008.508] 	(Run 'mkfontdir' on "/usr/share/fonts/100dpi/").
[  1008.508] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/75dpi/".
[  1008.508] 	Entry deleted from font path.
[  1008.509] 	(Run 'mkfontdir' on "/usr/share/fonts/75dpi/").
[  1008.509] (==) FontPath set to:
	/usr/share/fonts/misc/,
	/usr/share/fonts/TTF/,
	/usr/share/fonts/OTF/
[  1008.509] (==) ModulePath set to "/usr/lib/xorg/modules"
[  1008.509] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[  1008.509] (II) Loader magic: 0x821d60
[  1008.509] (II) Module ABI versions:
[  1008.509] 	X.Org ANSI C Emulation: 0.4
[  1008.509] 	X.Org Video Driver: 23.0
[  1008.509] 	X.Org XInput driver : 24.1
[  1008.509] 	X.Org Server Extension : 10.0
[  1008.510] (++) using VT number 1

[  1008.510] (--) controlling tty is VT number 1, auto-enabling KeepTty
[  1008.512] (II) systemd-logind: took control of session /org/freedesktop/login1/session/c1
[  1008.513] (II) xfree86: Adding drm device (/dev/dri/card0)
[  1008.514] (II) systemd-logind: got fd for /dev/dri/card0 226:0 fd 11 paused 0
[  1008.515] (--) PCI:*(0:0:2:0) 8086:0166:106b:0102 rev 9, Mem @ 0xa0000000/4194304, 0x90000000/268435456, I/O @ 0x00002000/64, BIOS @ 0x????????/131072
[  1008.515] (WW) Open ACPI failed (/var/run/acpid.socket) (No such file or directory)
[  1008.515] (II) LoadModule: "glx"
[  1008.515] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[  1008.516] (II) Module glx: vendor="X.Org Foundation"
[  1008.516] 	compiled for 1.19.1, module version = 1.0.0
[  1008.516] 	ABI class: X.Org Server Extension, version 10.0
[  1008.516] (==) Matched intel as autoconfigured driver 0
[  1008.516] (==) Matched intel as autoconfigured driver 1
[  1008.516] (==) Matched modesetting as autoconfigured driver 2
[  1008.516] (==) Matched fbdev as autoconfigured driver 3
[  1008.516] (==) Matched vesa as autoconfigured driver 4
[  1008.516] (==) Assigned the driver to the xf86ConfigLayout
[  1008.516] (II) LoadModule: "intel"
[... 259 lines cut ...]
[  1008.646] (II) UnloadModule: "mtrack"
[  1008.646] (II) config/udev: Adding input device applesmc (/dev/input/event6)
[  1008.646] (II) No input driver specified, ignoring this device.
[  1008.646] (II) This device may have been added with another device file.
[  1008.646] (II) config/udev: Adding input device applesmc (/dev/input/js0)
[  1008.646] (II) No input driver specified, ignoring this device.
[  1008.646] (II) This device may have been added with another device file.
[  1008.647] (II) config/udev: Adding input device PC Speaker (/dev/input/event4)
[  1008.647] (II) No input driver specified, ignoring this device.
[  1008.647] (II) This device may have been added with another device file.
[  1008.815] (II) intel(0): EDID vendor "APP", prod id 40980
[  1008.815] (II) intel(0): Printing DDC gathered Modelines:
[  1008.815] (II) intel(0): Modeline "2560x1600"x0.0  268.50  2560 2608 2640 2720  1600 1603 1609 1646 +hsync -vsync (98.7 kHz eP)
[  1031.988] (EE) kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.5.4/touchpad_jumping_cursor.html for details
[  1032.635] (EE) kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.5.4/touchpad_jumping_cursor.html for details
[  1032.796] (EE) kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.5.4/touchpad_jumping_cursor.html for details
[  1032.915] (EE) kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.5.4/touchpad_jumping_cursor.html for details
[  1069.312] (EE) kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.5.4/touchpad_jumping_cursor.html for details
[  1080.146] (EE) kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.5.4/touchpad_jumping_cursor.html for details

p2rkw, 2017-01-16:
It's not related to original issue.

In your configuration libinput 'takes over' touchpad, while mtrack gets /dev/input/mouse0 device. 
When did use BlueDragonX version last time? I think it's rather problem with default configuration of libinput: is 'steals' touchpad device (/dev/input/event12), even if you configured your system differently.
First try to force mtrack to use /dev/input/event12, to confirm that, and then raise a ticket in arch or libinput bugtracker. Probably arch udev rules have too high priority, or sth, anyway I think your custom configuration should overwrite rules added by udev. Please raise tickets there, and link them here.

lewishsh, 2017-01-16:
Okay. Sorry about that, and thanks for the help.

I'm fairly new to this: can you point me in the right direction re: how to force mtrack to use event12?

p2rkw, 2017-01-16:
Try something like this: https://github.com/p2rkw/xf86-input-mtrack/issues/11#issuecomment-216642566
Put it into /etc/X11/xorg.conf.d/50-mtrack.conf or /etc/X11/xorg.conf
If you installed mtrack via package manager (aur or something) ask package maintainer, because additional steps after installation shouldn't be required to make it work.

lewishsh, 2017-01-16:
Strange. I

lewishsh, 2017-01-16:
Thanks. I got it to work, not by forcing mtrack to use /dev/input/event12, but by moving my config from /usr/share/X11/xorg.conf.d to /etc/X11/xorg.conf.d. I'm not quite sure why that made a difference, and somehow it installs into /usr/share/X11/xorg.conf.d in the first place.

patarapolw, 2020-09-15:
It seems to work? but I get repeated left clicks, rather three-finger drag.

Xubuntu 20.04 on MacBook Pro 13" mid 2012