p2rkw/xf86-input-mtrack #23: Vertical movement with low sensitivity is ignored

paulthomson, 2016-10-05:
1. Set the sensitivity to a low value (e.g. 0.2). 
2. Very slowly move a finger from the top-left of the trackpad to the bottom-right. Vary the speed to see the issue. At a low enough speed, the cursor will only move horizontally. 

Alternatively:
1. Very slowly move a finger straight down. The cursor will not move at all unless you reach a certain speed. It is not possible to do this horizontally.

In both cases, the cursor appears to almost jump down in 10-20 pixel "steps" when moving slow enough. This makes it very difficult to make small movements. 

```
$ xinput --list-props 11
Device 'bcm5974':
	Device Enabled (139):	1
	Coordinate Transformation Matrix (141):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (272):	0
	Device Accel Constant Deceleration (273):	1.000000
	Device Accel Adaptive Deceleration (274):	1.000000
	Device Accel Velocity Scaling (275):	10.000000
	Trackpad Disable Input (276):	0
	Trackpad Sensitivity (277):	0.200000
	Trackpad Touch Pressure (278):	10, 10
	Trackpad Button Settings (279):	1, 1
	Trackpad Button Emulation Settings (280):	0, 1, 100
	Trackpad Button Emulation Values (281):	3, 2, 0
	Trackpad Tap Settings (282):	50, 120, 400
	Trackpad Tap Button Emulation (283):	0, 0, 0, 0
	Trackpad Thumb Detection (284):	0, 0
	Trackpad Thumb Size (285):	25, 70
	Trackpad Palm Detection (286):	1, 1
	Trackpad Palm Size (287):	40
	Trackpad Gesture Settings (288):	10, 100
	Trackpad Smooth Scroll (289):	1
	Trackpad Scroll Settings (290):	250, 20, 0
	Trackpad Scroll Buttons (291):	5, 4, 7, 6
	Trackpad Swipe Settings (292):	700, 300, 0
	Trackpad Swipe Buttons (293):	0, 0, 8, 9
	Trackpad Swipe4 Settings (294):	700, 300, 0
	Trackpad Swipe4 Buttons (295):	0, 0, 0, 0
	Trackpad Scroll Coasting (296):	0.100000, 200.000000
	Trackpad Scale Distance (297):	150
	Trackpad Scale Buttons (298):	12, 13
	Trackpad Rotate Distance (299):	150
	Trackpad Rotate Buttons (300):	14, 15
	Trackpad Hold1Move1 Stationary Settings (301):	20, 1
	Trackpad Hold1Move1 Settings (302):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (303):	1, 1, 1, 1
	Trackpad Drag Settings (304):	1, 350, 40, 200
	Trackpad Axis Inversion (305):	0, 0
	Trackpad Edge Size (306):	10
```

paulthomson, 2016-10-05:
Actually, it looks like it might be possible to also do this horizontally, so perhaps it is related to the aspect ratio of the screen, and so is just an artifact of having a low sensitivity. Still, with a low enough sensitivity, it is possible to slowly move your finger all the way across the trackpad without the cursor moving; I guess movements that are deemed "too small" since the last update are ignored. If this is the case, is there a way to adjust this?

p2rkw, 2016-10-09:
Fixed.

closed 2016-10-09 by p2rkw