p2rkw/xf86-input-mtrack #25: Feature request: wait time after click to allow for movement again.

mcourteaux, 2016-11-02:
I have a MacBook Pro. Driver works great so far (thanks for your work!).

When I click the touchpad (physically clicking the touchpad as a whole), it is often accompanied with the cursor jumping a little around. After a couple of minutes testing around when it happens, I guess that this would be fixed by two new wait time options:

 - The time cursor movement is disabled after a click-down.
 - The time a tap should exist (so the time the finger should be on the touchpad) before that tap is allowed to start moving the cursor. This should be typically a very short time (I'm guessing only a couple of milliseconds).

However, I'm not sure if those wait times would be needed if the driver support speed like in OS X. There, increasing the speed of a finger causes a super-linear speedup of the cursor. So basically: moving the finger slowly allows for more precision.

(I just switched to Linux, because of some problems I had. I already tweaked this driver before, but I'm slammed with work for school right now. If find some time, I could do it as well, but before I start, I wanted to contact you. Especially: what do you suggest to do? Will you do it? If not, but are willing to accept a Pull Request, any guidelines you want me to follow?)

p2rkw, 2016-11-03:
There are already settings "ButtonTouchExpire" and "FingerHigh", at first I'd suggest modifying them: you can do this by editing your xorg.conf, or using xinput to tweak driver without having to restart Xorg.
If you decide to use xinput, here are properties explained: https://github.com/p2rkw/xf86-input-mtrack/blob/master/include/mprops.h#L38
I hope it can be done without additional code, anyway I will gladly accept PR with example configuration for OSX devices.

PS.
There are also "MaxTapTime" and "GestureWaitTime", so there are plenty of options to try.

p2rkw, 2016-11-07:
PSS. Once all configuration possibilities listed above fail, we can discuss possible solutions, and how to modify source code. 
It's difficult to tweak driver for apple trackpad device without having one.

mcourteaux, 2016-11-22:
Just letting you know. I installed macOS again next to Linux. And I'm using that now most of the time again, so the need is less to get this fixed for me now. Moreover, in Linux I switched back to the default driver shipped with Ubuntu, since this driver is somehow incompatible with the "push cursor beyond screen boundaries" feature, required to make the Ubuntu dock-thingy popup when having auto-hide enabled. I highly suspected this to be related to xf86, since I couldn't find anything that causes this behaviour in your code.

I found this, and commented (completely on the bottom), however, no response so far: https://bugs.launchpad.net/ubuntu/+source/unity/+bug/938758