p2rkw/xf86-input-mtrack #26: Clickpad clicks always register as left-click and three finger swipe registers as two finger

beojan, 2016-11-05:
Clickpad clicks are always registering as left-clicks, and three finger swipes are registering as two finger scrolls. 

```
Device 'SynPS/2 Synaptics TouchPad':
	Device Enabled (138):	1
	Coordinate Transformation Matrix (140):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (268):	0
	Device Accel Constant Deceleration (269):	1.000000
	Device Accel Adaptive Deceleration (270):	1.000000
	Device Accel Velocity Scaling (271):	10.000000
	Trackpad Disable Input (300):	0
	Trackpad Sensitivity (301):	0.500000
	Trackpad Touch Pressure (302):	5, 5
	Trackpad Button Settings (303):	1, 1
	Trackpad Button Emulation Settings (304):	0, 1, 100
	Trackpad Button Emulation Values (305):	1, 3, 2
	Trackpad Tap Settings (306):	50, 120, 400
	Trackpad Tap Button Emulation (307):	1, 3, 0, 0
	Trackpad Thumb Detection (308):	0, 0
	Trackpad Thumb Size (309):	25, 70
	Trackpad Palm Detection (310):	1, 0
	Trackpad Palm Size (311):	40
	Trackpad Gesture Settings (312):	10, 100
	Trackpad Smooth Scroll (313):	1
	Trackpad Scroll Settings (314):	150, 20, 0
	Trackpad Scroll Buttons (315):	5, 4, 7, 6
	Trackpad Swipe Settings (316):	700, 300, 0
	Trackpad Swipe Buttons (317):	8, 9, 10, 11
	Trackpad Swipe4 Settings (318):	700, 300, 0
	Trackpad Swipe4 Buttons (319):	0, 0, 0, 0
	Trackpad Scroll Coasting (320):	0.100000, 200.000000
	Trackpad Scale Distance (321):	150
	Trackpad Scale Buttons (322):	12, 13
	Trackpad Rotate Distance (323):	150
	Trackpad Rotate Buttons (324):	14, 15
	Trackpad Hold1Move1 Stationary Settings (325):	20, 1
	Trackpad Hold1Move1 Settings (326):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (327):	1, 1, 1, 1
	Trackpad Drag Settings (328):	1, 350, 40, 200
	Trackpad Axis Inversion (329):	0, 0
	Trackpad Edge Size (330):	10
```

```
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier "Touchpads"
    Driver "mtrack"
    Option "Sensitivity" "0.5"
    Option "IgnorePalm" "on"
    Option "ButtonIntegrated" "on"
    Option "ClickFinger1" "1"
    Option "ClickFinger2" "3"
    Option "ClickFinger3" "2"
    Option "TapButton1" "1"
    Option "TapButton2" "3"
    Option "TapButton3" "2"
    Option "ScrollUpButton" "5"
    Option "ScrollDownButton" "4"
    Option "ScrollLeftButton" "7"
    Option "ScrollRightButton" "6"
    Option "ButtonTouchExpire" "0"
    Option "ScrollSmooth" "on"
    Option "SwipeDistance" "150"

EndSection
```

p2rkw, 2016-11-06:
Ad. clickpad behavior: you should enable button zones with
`
Option "ButtonZonesEnable" "true"
`

Ad. swipes: I appled your setup, but I couldn't reproduce, sorry. Does it happen all the time, or just accidentally?

beojan, 2016-11-06:
> ` Option "ButtonZonesEnable" "true"`

Is this neccessary for multi finger clicks, as opposed to clicks in the different areas of the touchpad?

> Ad. swipes: I appled your setup, but I couldn't reproduce, sorry. Does it happen all the time, or just accidentally?

All the time.

p2rkw, 2016-11-06:
>     Option "ButtonZonesEnable" "true"
> 
> Is this neccessary for multi finger clicks, as opposed to clicks in the different areas of the touchpad?
No, it shouldn't be necessary to turn it on for clicks with multiple fingers.

p2rkw, 2016-11-06:
So first attempt to bugfix is here: https://github.com/p2rkw/xf86-input-mtrack/tree/button_emulated_fix
Please test it with xinput --test <dev id>

beojan, 2016-11-10:
That's fixed two-finger clicks but not three finger (though three finger taps don't work either). Both register as a right (two finger) click, so perhaps this is connected to the swipe issue.

By the way, three and four finger swipes work correctly under Windows with the Synaptics driver, so the hardware definitely does support these.

p2rkw, 2016-11-12:
You set in xorg.conf: `Option "SwipeDistance" "150"`, but xinput shows: `Trackpad Swipe Settings (316):	700, 300, 0`, it should be 150 instead of 700. Could you attach/paste your Xorg.0.log?

beojan, 2016-11-13:
It doesn't anymore:

```
Device 'SynPS/2 Synaptics TouchPad':
        Device Enabled (138):   1
        Coordinate Transformation Matrix (140): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (268):     0
        Device Accel Constant Deceleration (269):       1.000000
        Device Accel Adaptive Deceleration (270):       1.000000
        Device Accel Velocity Scaling (271):    10.000000
        Trackpad Disable Input (300):   0
        Trackpad Sensitivity (301):     0.550000
        Trackpad Touch Pressure (302):  0, 0
        Trackpad Button Settings (303): 1, 1
        Trackpad Button Emulation Settings (304):       0, 1, 0
        Trackpad Button Emulation Values (305): 1, 3, 2
        Trackpad Tap Settings (306):    50, 150, 400
        Trackpad Tap Button Emulation (307):    1, 3, 2, 0
        Trackpad Thumb Detection (308): 0, 0
        Trackpad Thumb Size (309):      25, 70
        Trackpad Palm Detection (310):  1, 0
        Trackpad Palm Size (311):       40
        Trackpad Gesture Settings (312):        20, 100
        Trackpad Smooth Scroll (313):   1
        Trackpad Scroll Settings (314): 150, 20, 0
        Trackpad Scroll Buttons (315):  5, 4, 7, 6
        Trackpad Swipe Settings (316):  150, 300, 0
        Trackpad Swipe Buttons (317):   8, 9, 10, 11
        Trackpad Swipe4 Settings (318): 700, 300, 0
        Trackpad Swipe4 Buttons (319):  0, 0, 0, 0
        Trackpad Scroll Coasting (320): 0.100000, 200.000000
        Trackpad Scale Distance (321):  150
        Trackpad Scale Buttons (322):   12, 13
        Trackpad Rotate Distance (323): 150
        Trackpad Rotate Buttons (324):  14, 15
        Trackpad Hold1Move1 Stationary Settings (325):  20, 1
        Trackpad Hold1Move1 Settings (326):     1, 0, 1000
        Trackpad Hold1Move1 Buttons (327):      1, 1, 1, 1
        Trackpad Drag Settings (328):   1, 350, 40, 200
        Trackpad Axis Inversion (329):  0, 0
        Trackpad Edge Size (330):       10
```

Swipe still doesn't work correctly.

p2rkw, 2016-11-14:
I just checked your configuration with xinput --test <dev id> and I couldn't reproduce on my device.
Could you put some debug printfs(`LOG_INFO("your message")`) into code and share your results?

p2rkw, 2016-12-02:
Any problems with last fix? Can I merge it to master?

beojan, 2016-12-02:
I actually gave up and switched to libinput. Sorry.

p2rkw, 2017-12-17:
Original issue may be related to #52: right now there's no support for [partial multi-touch touchpads](https://wayland.freedesktop.org/libinput/doc/1.2.1/touchpads.html). I will try to add it somehow.