p2rkw/xf86-input-mtrack #27: Two finger scroll randomly selects texts and clicks

martinkronstad, 2016-12-29:
While scrolling with two fingers, the trackpad randomly clicks links and selects texts. It does now happen all the time, just ranomly.

I have tried to remove all tap features, since I do not need it.

Here is my config file:

`Section "InputClass"
    MatchIsTouchpad "on"
    Identifier "Touchpads"
    Driver "mtrack"
    Option "IgnoreThumb" "true"
    Option "ThumbSize" "15"
    Option "IgnorePalm" "true"
    Option "ScrollDistance" "25"
    Option "ClickFinger1" "1"
    Option "ClickFinger2" "3"
    Option "ClickFinger3" "0"
    Option "EdgeSize" "17"
    Option "TapDragEnable" "false"
    Option "TapButton1" "0"
    Option "TapButton2" "0"
    Option "TapButton3" "0"
    Option "Hold1Move1StationaryButton" "0"
    Option "TapButton4" "0"
    Option "ScrollCoastDuration" "400"
    Option "ScrollLeftButton" "0"
    Option "ScrollRightButton" "0"
    Option "ButtonZonesEnable" "1"
EndSection

`

p2rkw, 2016-12-29:
Is it corelated with distance between fingers?

martinkronstad, 2016-12-29:
Not sure what you mean, but It should not tap at all since I have tried to disable all tapping? I keep my fingers pretty much together while scrolling, maybe I need to ajust something? The trackpad may be detecting it as one finger??

martinkronstad, 2016-12-31:
I actually figured this out. By setting TapDragTime to a high number (I set mine to 40000), the problem disappears. So the thing was that I has holding my fingers on the trackpad for too long before moving them. That triggered the TapDrag.

closed 2016-12-31 by martinkronstad