p2rkw/xf86-input-mtrack #28: Config request: Add option to only set BottomEdge size

martinkronstad, 2016-12-31:
Right now there is only an option to set the edge for the whole trackpad, since I like to rest my thumb on the trackpad, or hold a finger on the button area, I need to set the Edge-option. 

I have the Edge-option at 15%, this causes me to miss my scroll area on the trackpad from time to time. I Would like to set my Edge property to 15-20% for the bottom edge, and 0 for the other sides.

p2rkw, 2017-01-13:
Implementation would be very easy, just check [1] and [2], but I'm sure how to deal with configuration options name change. I think 'EdgeSize' should set size for all edges, and 'BottomEdgeSize' should overwrite only bottom value.

[1] https://github.com/p2rkw/xf86-input-mtrack/blob/eb3b9c3219603d48c8792b7bdb99ddc3040c6320/src/mconfig.c#L191
[2] https://github.com/p2rkw/xf86-input-mtrack/blob/eb3b9c3219603d48c8792b7bdb99ddc3040c6320/src/mtstate.c#L132

ericfreese, 2017-08-24:
Could also break it out into separate options: `EdgeSizeTop`, `EdgeSizeLeft`, `EdgeSizeBottom`, `EdgeSizeRight`. Though that brings up the question of whether you'd keep the `EdgeSize` option for backwards compatibility.

p2rkw, 2017-08-24:
Please test: https://github.com/p2rkw/xf86-input-mtrack/commit/8f73751c80ee1e164b03a8886e871b8056b23405

ericfreese, 2017-08-25:
The separate options do seem to work on 8f73751, but I get seem to get some funky behavior on my machine. I have a macbook pro with an integrated touchpad. To select text, I have my right index finger touching on the center of the touchpad to position the pointer. I then add my thumb at the bottom of the touchpad (within the EdgeSize bounds) and press the button down. While holding the button down with my thumb I move the pointer with my index finger. This operation works fine in e2a9c29, but is broken with the next commit 3aa9b09 (once I push the button down with my thumb the pointer stops moving when I move my index finger).

ericfreese, 2017-08-25:
I can confirm that rebasing f27a2e8 on master (41d646e) solves the issue for me without introducing any weird behavior.

p2rkw, 2017-08-27:
Thanks for tracking it down, I added edge scrolling and few other changes, on [edge_scroll](https://github.com/p2rkw/xf86-input-mtrack/tree/edge_scroll) branch.

Please report found issues.

p2rkw, 2017-10-28:
Done, it's on master.

closed 2017-10-28 by p2rkw