p2rkw/xf86-input-mtrack #34: `modprobe -r hid_apple && modprobe hid_apple` breaks driver, not sure how to restore

rmanne, 2017-04-19:
Possibly due to some other bug, after resuming from hibernation, one of two things happen:

1. Resume fails due to some inconsistency (not sure how to debug, unrelated)

2. Resume succeeds at which point, trackpad is working just fine, but as soon as I press any key on my keyboard, xorg crashes. Interestingly enough, I found that if I `modprobe -r hid_apple && modprobe hid_apple`, it stops crashing in this case.

BUT, if I run `modprobe -r hid_apple && modprobe hid_apple`, trackpad stops working.

Oddly, the result of running xinput indexes between the keyboard and the trackpad swap places (eg: initially keyboard at 11 and trackpad at 12, after it becomes keyboard at 12 and trackpad at 11). This interaction (as well as list-props on the trackpad field is attached below).

[image]

How do I restore the trackpad after `modprobe -r hid_apple && modprobe hid_apple`, which appears to be essential for hibernation?

p2rkw, 2017-04-20:
Hi,
as a bandaid fix you could try 'solution' from this comment: https://unix.stackexchange.com/questions/37266/reload-xserver-xorg-input-mouse#comment118756_37266

rmanne, 2017-04-20:
Initially, the trackpad seems to be working just fine. I don't think it gets unloaded during hibernation. But it presumably gets unloaded if you `modprobe -r hid_apple`. Xorg crashes only with the keyboard.

If you have a macbook, can you tell me how you set up hibernation? I'm using swsusp. I tried uswsusp but it did not work at all.

p2rkw, 2017-04-21:
Sorry if I wasn't precise: I don't have mac laptop, but dell, but I'm using hibernation just fine.
Here are some hints from arch wiki:
https://wiki.archlinux.org/index.php/Mac#Suspend_.26_Power_Off_.2811.2C4.2B.29
https://wiki.archlinux.org/index.php/Mac#Suspend_and_Hibernate

rmanne, 2017-04-21:
I checked it out, but it looks like there patched kernel `linux-macbook` is for the newer Macbook Pros that can't even shutdown properly. I'm on a Macbook Air 2013. For me, everything except hibernation works. I'll try it anyways and see if it works.

rmanne, 2017-07-01:
I'd just assume that there's something very wrong with hibernating on a macbook. I'm not going to actively look for a solution to this, so I'll close it.

closed 2017-07-01 by rmanne

p2rkw, 2017-08-20:
Please try with this: https://github.com/p2rkw/xf86-input-mtrack/commit/a898c4aa5e6505fd5abfc80966e392d20d8e4b9b commit (branch https://github.com/p2rkw/xf86-input-mtrack/tree/edge_scroll).