p2rkw/xf86-input-mtrack #35: Tapping enabled when multiple fingers are on touchpad

1k2ka, 2017-05-13:
When 2 or more fingers are on the touchpad tapping gets enabled. It's reproducible by holding one finger on the touchpad and tapping with another finger.

```
Device 'Apple Inc. Apple Internal Keyboard / Trackpad':
	Device Enabled (139):	1
	Coordinate Transformation Matrix (141):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Send Events Modes Available (261):	1, 0
	libinput Send Events Mode Enabled (262):	0, 0
	libinput Send Events Mode Enabled Default (263):	0, 0
	Device Node (264):	"/dev/input/event12"
	Device Product ID (265):	1452, 627
```

```
Section "InputClass"
        MatchIsTouchpad "on"
        Identifier      "Touchpads"
        Driver          "mtrack"
        Option          "Sensitivity" "0.65"
        Option          "FingerHigh" "5"
        Option          "FingerLow" "1"
        Option          "IgnoreThumb" "true"
        Option          "ThumbRatio" "70"
        Option          "ThumbSize" "25"
        Option          "IgnorePalm" "true"
        Option          "TapButton1" "0"
        Option          "TapButton2" "0"
        Option          "TapButton3" "0"
        Option          "TapButton4" "0"
        Option          "ClickFinger1" "3"
        Option          "ClickFinger2" "3"
        Option          "ClickFinger3" "3"
        Option          "ButtonMoveEmulate" "false"
        Option          "ButtonIntegrated" "true"
        Option          "ClickTime" "25"
        Option          "BottomEdge" "30"
        Option          "ScrollCoastDuration" "500"
        Option          "ScrollCoastEnableSpeed" ".3"
        Option          "ScrollUpButton" "5"
        Option          "ScrollDownButton" "4"
        Option          "ScrollLeftButton" "6"
        Option          "ScrollRightButton" "7"
        Option          "ScrollDistance" "250"
	Option          "TapDragEnable" "false"
	Option		"TapDragWait" "40000"
EndSection

```

p2rkw, 2017-05-15:
Are you sure you are using mtrack? First log shows libinput was selected. Please attach xinput output for device controlled by mtrack, and preferably attach your xorg.log

1k2ka, 2017-05-15:
Making changes in my config file affects the touchpad at least. According to the log mtrack loads but fails configuring the device. This should be the proper xinput log I think:

```
Device 'bcm5974':
	Device Enabled (139):	1
	Coordinate Transformation Matrix (141):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (272):	0
	Device Accel Constant Deceleration (273):	1.000000
	Device Accel Adaptive Deceleration (274):	1.000000
	Device Accel Velocity Scaling (275):	10.000000
	Trackpad Disable Input (302):	0
	Trackpad Sensitivity (303):	0.650000
	Trackpad Touch Pressure (304):	5, 1
	Trackpad Button Settings (305):	1, 1
	Trackpad Button Emulation Settings (306):	0, 0, 100
	Trackpad Button Emulation Values (307):	3, 3, 3
	Trackpad Tap Settings (308):	25, 120, 400
	Trackpad Tap Button Emulation (309):	0, 0, 0, 0
	Trackpad Thumb Detection (310):	1, 0
	Trackpad Thumb Size (311):	25, 70
	Trackpad Palm Detection (312):	1, 0
	Trackpad Palm Size (313):	40
	Trackpad Gesture Settings (314):	10, 100
	Trackpad Smooth Scroll (315):	1
	Trackpad Scroll Settings (316):	250, 20, 0
	Trackpad Scroll Buttons (317):	5, 4, 7, 6
	Trackpad Swipe Settings (318):	700, 300, 0
	Trackpad Swipe Buttons (319):	8, 9, 10, 11
	Trackpad Swipe4 Settings (320):	700, 300, 0
	Trackpad Swipe4 Buttons (321):	0, 0, 0, 0
	Trackpad Scroll Coasting (322):	0.300000, 500.000000
	Trackpad Scale Distance (323):	150
	Trackpad Scale Buttons (324):	12, 13
	Trackpad Rotate Distance (325):	150
	Trackpad Rotate Buttons (326):	14, 15
	Trackpad Hold1Move1 Stationary Settings (327):	20, 1
	Trackpad Hold1Move1 Settings (328):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (329):	1, 1, 1, 1
	Trackpad Drag Settings (330):	0, 350, 40000, 200
	Trackpad Axis Inversion (331):	0, 0
	Trackpad Edge Size (332):	10
```

```
> cat .local/share/xorg/Xorg.0.log | grep mtrack
[ 23573.210] (II) LoadModule: "mtrack"
[ 23573.210] (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
[ 23573.210] (II) Module mtrack: vendor="X.Org Foundation"
[ 23573.210] (II) Using input driver 'mtrack' for 'bcm5974'
[ 23573.211] (II) mtrack: devname: bcm5974
[ 23573.211] (II) mtrack: devid: 5ac 273 1
[ 23573.211] (II) mtrack: caps: left mtdata ibt
[ 23573.211] (II) mtrack: 0: min: 0 max: 2048
[ 23573.211] (II) mtrack: 1: min: 0 max: 2048
[ 23573.211] (II) mtrack: 2: min: 0 max: 2048
[ 23573.211] (II) mtrack: 3: min: 0 max: 2048
[ 23573.211] (II) mtrack: 4: min: -16384 max: 16384
[ 23573.211] (II) mtrack: 5: min: -4828 max: 5345
[ 23573.211] (II) mtrack: 6: min: -203 max: 6803
[ 23573.211] (II) mtrack: 9: min: 0 max: 65535
[ 23573.252] (II) Using input driver 'mtrack' for 'bcm5974'
[ 23573.253] (EE) mtrack: cannot configure device
[ 23573.253] (II) UnloadModule: "mtrack"
```
Full logs: https://gist.github.com/1k2ka/ae0471e9b26f7875478688ca2c91bfa4

p2rkw, 2017-05-16:
It may be caused by my "feature" Hold1Move1, first try to disable it with:
```
Option "Hold1Move1StationaryButton" "0"
```
in proper xorg.conf section

1k2ka, 2017-05-16:
Adding `Option "Hold1Move1StationaryButton" "0"` to my config seems to have fixed the issue. This is the full input of `xinput` if you're still interested.

```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Broadcom Corp. Bluetooth USB Host Controller	id=11	[slave  pointer  (2)]
⎜   ↳ bcm5974                                 	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Broadcom Corp. Bluetooth USB Host Controller	id=10	[slave  keyboard (3)]
    ↳ Apple Inc. Apple Internal Keyboard / Trackpad	id=12	[slave  keyboard (3)]

```

p2rkw, 2017-05-17:
Example updated: https://github.com/p2rkw/xf86-input-mtrack/commit/02f458ac3203ef3e18e607a6a60f7580012522a4

closed 2017-05-17 by p2rkw