p2rkw/xf86-input-mtrack #36: Button Zones do not work when thumb is resting on trackpad

es92, 2017-05-18:
I have Button Zones configured as follows (full config [here](https://github.com/p2rkw/xf86-input-mtrack/files/1009470/10-mtrack.conf.txt))

```
    Option "ClickFinger1" "1"
    Option "ClickFinger2" "1"
    Option "ClickFinger3" "3"
```
The thirds work as expected when clicking with one finger (right third does right click)

When resting a finger on the top half of the left third of the trackpad, clicking on the right third of the trackpad does perform a right click as expected.

However, when a finger is resting on the bottom half of the left third of the trackpad, clicking on the right third performs a left click.

Thanks,
Evan

xavierhd, 2018-06-06:
About that, better late than never. It is because the click event is binded to the lowest valid touch.

xavierhd, 2018-06-06:
@es92 if you still exist, can you try this patch 5fba330?

xavierhd, 2018-06-06:
@p2rkw Is there a reason the current behavior is used?

p2rkw, 2018-06-07:
@xavierhd : https://github.com/p2rkw/xf86-input-mtrack/commit/af4167651c392cd2dfc7ecdf7e40ceff6bbc72f6

So it was clearly done on purpose.

xavierhd, 2018-06-19:
You and they where right, i tested the behavior using the 5fba330 patch...
It is awful to use :disappointed: 
You should close this.

closed 2018-12-06 by p2rkw