p2rkw/xf86-input-mtrack #37: Support for xorg-server 1.19.3.-2?

BurningSmile, 2017-05-20:
I'm trying to install this on my arch based laptop. Any chance of bumping the supported xorg version?

p2rkw, 2017-05-21:
As long as Xorg dodn't break ABI you should be able to compile and install mtrack just fine. 
Do you have some error message from compilation or installation?

BurningSmile, 2017-05-21:
configure: error: Package requirements (xorg-server >= 1.7 xproto inputproto ) were not met:

No package 'xorg-server' found

tried with both ./configure and 
./configure --with-xorg-module-dir=/usr/lib/xorg/modules

From here

https://howchoo.com/g/mdy0ngziogm/the-perfect-almost-touchpad-settings-on-linux-2

p2rkw, 2017-05-22:
Did you tried ```sudo apt build-dep xserver-xorg-input-mtrack``` beforehand? Looks looks you don't have ```xorg-server-dev``` installed. Install & reconfigure.

BurningSmile, 2017-05-22:
That was it, thanks!

closed 2017-05-22 by BurningSmile

subdavis, 2018-10-12:
Could we put this in the readme?  It would help a lot of people...