p2rkw/xf86-input-mtrack #38: crashes xorg in ubuntu 17.04

vladie, 2017-06-03:
i've compiled module for ubuntu 17.04
and tried to use it on my dell laptop with apple trackpad

it works, but crashes x server definitely in few minutes after start
backtrace are different each time, so it seems to be memory corruption

vladie, 2017-06-03:
This is props of my trackpad:

Device 'Work’s Trackpad':
	Device Enabled (153):	1
	Coordinate Transformation Matrix (155):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (283):	0
	Device Accel Constant Deceleration (284):	1.000000
	Device Accel Adaptive Deceleration (285):	1.000000
	Device Accel Velocity Scaling (286):	10.000000
	Trackpad Disable Input (583):	0
	Trackpad Sensitivity (584):	1.000000
	Trackpad Touch Pressure (585):	5, 5
	Trackpad Button Settings (586):	1, 1
	Trackpad Button Emulation Settings (587):	0, 1, 100
	Trackpad Button Emulation Values (588):	3, 2, 0
	Trackpad Tap Settings (589):	50, 120, 400
	Trackpad Tap Button Emulation (590):	1, 3, 2, 0
	Trackpad Thumb Detection (591):	0, 0
	Trackpad Thumb Size (592):	25, 70
	Trackpad Palm Detection (593):	0, 0
	Trackpad Palm Size (594):	40
	Trackpad Gesture Settings (595):	10, 100
	Trackpad Smooth Scroll (596):	1
	Trackpad Scroll Settings (597):	150, 20, 0
	Trackpad Scroll Buttons (598):	4, 5, 6, 7
	Trackpad Swipe Settings (599):	1, 0, 1000
	Trackpad Swipe Buttons (600):	1, 1, 1, 1
	Trackpad Swipe4 Settings (601):	700, 300, 0
	Trackpad Swipe4 Buttons (602):	0, 0, 0, 0
	Trackpad Scroll Coasting (603):	0.100000, 200.000000
	Trackpad Scale Distance (604):	150
	Trackpad Scale Buttons (605):	12, 13
	Trackpad Rotate Distance (606):	150
	Trackpad Rotate Buttons (607):	14, 15
	Trackpad Hold1Move1 Stationary Settings (608):	20, 1
	Trackpad Hold1Move1 Settings (609):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (610):	1, 1, 1, 1
	Trackpad Drag Settings (611):	1, 350, 40, 200
	Trackpad Axis Inversion (612):	0, 0
	Trackpad Edge Size (613):	10

p2rkw, 2017-06-06:
Where did you get backtraces from? xorg.log? dmesg? Could you paste/attach few examples?

p2rkw, 2017-08-20:
No reply. Close.

closed 2017-08-20 by p2rkw