p2rkw/xf86-input-mtrack #39: Сonfused vertical and horizontal scroll labels

alexbers, 2017-06-09:
Hi,

It seems that vertical and horizontal scroll labels are confused. It causes an unpredictable scrolling in applications that use Qt framework.

When mouse enters Qt window, QXcbConnection::updateScrollingDevice function is executed (http://code.qt.io/cgit/qt/qtbase.git/tree/src/plugins/platforms/xcb/qxcbconnection_xi2.cpp?h=5.6#n860). This function enumerates evaluators, looks for valuator with valuatorAtom == QXcbAtom::RelVertScroll label and updates scrollingDevice.lastScrollPosition. 

When labels are confused, the vertical scroll position is taken from valuator that tracks horizontal scroll position.

Here is a patch: https://github.com/alexbers/xf86-input-mtrack/commit/05b3176d6644cf93c9d7a8673063560427b5a3a1

p2rkw, 2017-06-11:
Picked. Thanks.

closed 2017-06-11 by p2rkw

p2rkw, 2017-06-11:
Your other commits look solid, too. I will merge them, once tested. I'm not sure how to handle 'EdgeSize' split in terms of stability for users.