p2rkw/xf86-input-mtrack #41: how can i get it work on mint 18.2

xigua5, 2017-07-13:
i have add the fellowing to /etc/X11/xorg.conf and my touchpad stop to work
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier      "Touchpads"
    Driver          "mtrack"
EndSection

i try to copy the examples macbook.conf to /etc/X11/xorg.conf and it still can not work
what can i do ?
thanks

xigua5, 2017-07-13:
`xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Lenovo USB Receiver                     	id=11	[slave  pointer  (2)]
⎜   ↳ Lenovo USB Receiver                     	id=12	[slave  pointer  (2)]
⎜   ↳ TPPS/2 IBM TrackPoint                   	id=15	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Lenovo USB Receiver                     	id=10	[slave  keyboard (3)]
    ↳ Integrated Camera                       	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                  	id=16	[slave  keyboard (3)]
`

`inxi -Fxzd
System:    Host: tbook Kernel: 4.8.0-53-generic x86_64 (64 bit gcc: 5.4.0)
           Desktop: Cinnamon 3.4.4 (Gtk 3.18.9-1ubuntu3.3)
           Distro: Linux Mint 18.2 Sonya
Machine:   System: LENOVO (portable) product: 20FWCTO1WW v: ThinkPad T460p
           Mobo: LENOVO model: 20FWCTO1WW v: SDK0J40700 WIN
           Bios: LENOVO v: R07ET65W (2.05 ) date: 04/27/2016
CPU:       Quad core Intel Core i7-6820HQ (-HT-MCP-) cache: 8192 KB
           flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx) bmips: 21696
           clock speeds: max: 3600 MHz 1: 799 MHz 2: 799 MHz 3: 799 MHz
           4: 799 MHz 5: 799 MHz 6: 799 MHz 7: 799 MHz 8: 799 MHz
Graphics:  Card-1: Intel Skylake Integrated Graphics bus-ID: 00:02.0
           Card-2: NVIDIA GM108M [GeForce 940MX] bus-ID: 02:00.0
           Display Server: X.Org 1.18.4 driver: nvidia
           Resolution: 1360x768@59.80hz
           GLX Renderer: GeForce 940MX/PCIe/SSE2
           GLX Version: 4.5.0 NVIDIA 375.66 Direct Rendering: Yes
Audio:     Card Intel Sunrise Point-H HD Audio
           driver: snd_hda_intel bus-ID: 00:1f.3
           Sound: Advanced Linux Sound Architecture v: k4.8.0-53-generic
Network:   Card-1: Intel Ethernet Connection (2) I219-LM
           driver: e1000e v: 3.2.6-k bus-ID: 00:1f.6
           IF: enp0s31f6 state: up speed: 1000 Mbps duplex: full mac: <filter>
           Card-2: Intel Wireless 8260 driver: iwlwifi bus-ID: 03:00.0
           IF: wlp3s0 state: down mac: <filter>
Drives:    HDD Total Size: 256.1GB (27.0% used)
           ID-1: /dev/sda model: PLEXTOR_PX size: 256.1GB
           Optical: No optical drives detected.
Partition: ID-1: / size: 232G used: 62G (28%) fs: ext4 dev: /dev/sda1
           ID-2: swap-1 size: 3.68GB used: 0.00GB (0%) fs: swap dev: /dev/sda5
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 44.0C mobo: N/A gpu: 0.0:45C
           Fan Speeds (in rpm): cpu: 2159
Info:      Processes: 271 Uptime: 1:21 Memory: 1578.7/19501.2MB
           Init: systemd runlevel: 5 Gcc sys: 5.4.0
           Client: Shell (bash 4.3.481) inxi: 2.2.35 
`

p2rkw, 2017-07-20:
Please attach your xorg.log. Probably libinput takes precedence over mtrack.

jl4c, 2017-08-07:
Hi, I have the same problem. Here is my log:
```
HP_ENVY% cat ~/.local/share/xorg/Xorg.0.log  
[  1251.174] 
X.Org X Server 1.19.3
Release Date: 2017-03-15
[  1251.174] X Protocol Version 11, Revision 0
[  1251.174] Build Operating System: Linux 4.9.11-1-ARCH x86_64 
[  1251.174] Current Operating System: Linux HP_ENVY 4.12.4-1-ARCH #1 SMP PREEMPT Fri Jul 28 18:54:18 UTC 2017 x86_64
[  1251.174] Kernel command line: vmlinuz-linux  root=UUID=9e56e0b9-a6f6-4f2c-98a2-9ce830b222b1 rw add_efi_memmap initrd=intel-ucode.img initrd=initramfs-linux.img 
[  1251.174] Build Date: 07 April 2017  05:42:48PM
[  1251.174]  
[  1251.174] Current version of pixman: 0.34.0
[  1251.174]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[  1251.174] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1251.175] (==) Log file: "/home/pepe/.local/share/xorg/Xorg.0.log", Time: Mon Aug  7 17:54:02 2017
[  1251.175] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  1251.175] (==) No Layout section.  Using the first Screen section.
[  1251.175] (==) No screen section available. Using defaults.
[  1251.175] (**) |-->Screen "Default Screen Section" (0)
[  1251.175] (**) |   |-->Monitor "<default monitor>"
[  1251.175] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[  1251.175] (==) Automatically adding devices
[  1251.175] (==) Automatically enabling devices
[  1251.175] (==) Automatically adding GPU devices
[  1251.175] (==) Automatically binding GPU devices
[  1251.175] (==) Max clients allowed: 256, resource mask: 0x1fffff
[  1251.175] (WW) The directory "/usr/share/fonts/Type1/" does not exist.
[  1251.175]    Entry deleted from font path.
[  1251.175] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/100dpi/".
[  1251.175]    Entry deleted from font path.
[  1251.175]    (Run 'mkfontdir' on "/usr/share/fonts/100dpi/").
[  1251.175] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/75dpi/".
[  1251.175]    Entry deleted from font path.
[  1251.175]    (Run 'mkfontdir' on "/usr/share/fonts/75dpi/").
[  1251.175] (==) FontPath set to:
        /usr/share/fonts/misc/,
        /usr/share/fonts/TTF/,
        /usr/share/fonts/OTF/
[  1251.175] (==) ModulePath set to "/usr/lib/xorg/modules"
[  1251.175] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[  1251.175] (II) Loader magic: 0x822d60
[  1251.175] (II) Module ABI versions:
[  1251.175]    X.Org ANSI C Emulation: 0.4
[  1251.175]    X.Org Video Driver: 23.0
[  1251.175]    X.Org XInput driver : 24.1
[  1251.175]    X.Org Server Extension : 10.0
[  1251.176] (++) using VT number 1

[  1251.176] (--) controlling tty is VT number 1, auto-enabling KeepTty
[  1251.178] (II) systemd-logind: took control of session /org/freedesktop/login1/session/c2
[  1251.178] (II) xfree86: Adding drm device (/dev/dri/card0)
[  1251.179] (II) systemd-logind: got fd for /dev/dri/card0 226:0 fd 11 paused 0
[  1251.182] (--) PCI:*(0:0:2:0) 8086:0416:103c:1967 rev 6, Mem @ 0xd3000000/4194304, 0xc0000000/268435456, I/O @ 0x00006000/64, BIOS @ 0x????????/131072
[  1251.182] (WW) Open ACPI failed (/var/run/acpid.socket) (No such file or directory)
[  1251.182] (II) LoadModule: "glx"
[  1251.182] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[  1251.183] (II) Module glx: vendor="X.Org Foundation"
[  1251.183]    compiled for 1.19.3, module version = 1.0.0
[  1251.183]    ABI class: X.Org Server Extension, version 10.0
[  1251.183] (==) Matched intel as autoconfigured driver 0
[  1251.183] (==) Matched intel as autoconfigured driver 1
[  1251.183] (==) Matched modesetting as autoconfigured driver 2
[  1251.183] (==) Matched fbdev as autoconfigured driver 3
[  1251.183] (==) Matched vesa as autoconfigured driver 4
[... 384 lines cut ...]
[  1251.426] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:046D:C52F.0002/input/input3/event2"
[  1251.426] (II) XINPUT: Adding extended input device "Logitech USB Receiver" (type: KEYBOARD, id 16)
[  1252.413] (II) modeset(0): EDID vendor "SDC", prod id 18001
[  1252.413] (II) modeset(0): Printing DDC gathered Modelines:
[  1252.413] (II) modeset(0): Modeline "1366x768"x0.0   76.30  1366 1430 1478 1606  768 770 775 792 -hsync -vsync (47.5 kHz eP)
[  1252.413] (II) modeset(0): Modeline "1366x768"x0.0   50.80  1366 1430 1478 1606  768 770 775 792 -hsync -vsync (31.6 kHz e)
[  1252.646] (II) modeset(0): EDID vendor "SDC", prod id 18001
[  1252.646] (II) modeset(0): Printing DDC gathered Modelines:
[  1252.646] (II) modeset(0): Modeline "1366x768"x0.0   76.30  1366 1430 1478 1606  768 770 775 792 -hsync -vsync (47.5 kHz eP)
[  1252.646] (II) modeset(0): Modeline "1366x768"x0.0   50.80  1366 1430 1478 1606  768 770 775 792 -hsync -vsync (31.6 kHz e)
[  3793.958] (II) config/udev: removing device Logitech USB Receiver
[  3793.958] (**) Option "fd" "25"
[  3793.958] (II) event1  - (II) Logitech USB Receiver: (II) device removed
[  3793.960] (II) UnloadModule: "libinput"
[  3793.960] (II) systemd-logind: releasing fd for 13:65
[  3794.028] (II) config/udev: removing device Logitech USB Receiver
[  3794.028] (**) Option "fd" "26"
[  3794.029] (II) UnloadModule: "libinput"
[  3794.029] (II) systemd-logind: not releasing fd for 13:66, still in use
[  3794.029] (II) config/udev: removing device Logitech USB Receiver
[  3794.029] (**) Option "fd" "26"
[  3794.029] (II) event2  - (II) Logitech USB Receiver: (II) device removed
[  3794.030] (II) UnloadModule: "libinput"
[  3794.030] (II) systemd-logind: releasing fd for 13:66
```

p2rkw, 2017-08-08:
https://github.com/p2rkw/xf86-input-mtrack/issues/21#issuecomment-272914476

closed 2017-10-28 by p2rkw