p2rkw/xf86-input-mtrack #42: Segfault after resuming from suspend on 0.4.1

m4rcu5, 2017-08-02:
After upgrading to 0.4.1, I encounter the following issue;
If I suspend my machine and wake it again xorg will segfault on the first keypress (not button press on trackpad)

```
[    35.114] (EE) Backtrace:
[    35.114] (EE) 0: /usr/bin/X (xorg_backtrace+0x4a) [0x591c2a]
[    35.114] (EE) 1: /usr/bin/X (0x400000+0x195949) [0x595949]
[    35.114] (EE) 2: /lib64/libpthread.so.0 (0x7fceb1ada000+0x10f40) [0x7fceb1aeaf40]
[    35.114] (EE) 3: /usr/bin/X (TimerSet+0xcb) [0x58f30b]
[    35.114] (EE) 4: /usr/bin/X (AccessXFilterPressEvent+0x242) [0x545622]
[    35.114] (EE) 5: /usr/bin/X (0x400000+0x174979) [0x574979]
[    35.115] (EE) 6: /usr/bin/X (mieqProcessInputEvents+0x99) [0x574ac9]
[    35.115] (EE) 7: /usr/bin/X (ProcessInputEvents+0x19) [0x477289]
[    35.115] (EE) 8: /usr/bin/X (0x400000+0x35886) [0x435886]
[    35.115] (EE) 9: /usr/bin/X (0x400000+0x39b08) [0x439b08]
[    35.115] (EE) 10: /lib64/libc.so.6 (__libc_start_main+0xf1) [0x7fceb17641e1]
[    35.115] (EE) 11: /usr/bin/X (_start+0x2a) [0x42377a]
[    35.115] (EE) 
[    35.115] (EE) Segmentation fault at address 0x8
[    35.115] (EE) 
Fatal server error:
[    35.115] (EE) Caught signal 11 (Segmentation fault). Server aborting
[    35.115] (EE) 
```

The interesting part is, that if i leave the system idle for about 10 minutes before closing the lid, it will wake without segfaulting on the first key press.

Downgrading to 0.3.1 will successfully mitigate the issue.

I've uploaded the full logs of a crashing session as gist: https://gist.github.com/m4rcu5/1c4f753e799a4f9fa3b55ecb8b55c42a

Any idea what could be going on here?

p2rkw, 2017-08-20:
Probably timer pointer is invalidated on wakeup, but it's used anyways after wake up which leads to crash.

p2rkw, 2017-08-20:
Please try with this: https://github.com/p2rkw/xf86-input-mtrack/commit/a898c4aa5e6505fd5abfc80966e392d20d8e4b9b commit (branch https://github.com/p2rkw/xf86-input-mtrack/tree/edge_scroll).

p2rkw, 2017-09-02:
Closing for now, reopen if still reproducible with current master.

closed 2017-09-02 by p2rkw

m4rcu5, 2017-10-22:
Confirmed that the latest master fixes the wakeup from sleep issue. Unfortunately the latest master breaks my left-click (opened issue #48 for that)