p2rkw/xf86-input-mtrack #43: High power consumption

hobarrera, 2017-08-15:
A few months back I moved from a 2013 MBA to a 2015 MBP. On the latter, the [force] touchpad is consuming about 10W, permanently. I'm wondering if this is something you've seen, or you have any idea if it might be related to the driver or something else.

Have you had a chance to test this specific hardware? Do you know of any way to determine if this very high power usage is normal, or driver-related?

As a reference, everything *aside* from the touchpad uses 4W (but the touchpad alone uses 10W). On the MBA, the touchpad used an insignificant amount of power, and the computer's total power usage was about 5W.

Let me know if I can provide anything that may help triage this, totally willing to help!

p2rkw, 2017-08-16:
How did you measure it? Powertop? I don't have access to MBP. Could you check consumption with synaptics and mtrack version v0.3.0? Do you observe increased power consumption all the time, or maybe only when scrolling? If scrolling is problematic please disable smooth scrolling and coasting and check power usage again. You can also check your Xorg.0.log for some debug logs - if extensive logging in on, it may increase power usage.

My initial thought is due to X11 timers, processor can't switch to idle mode. In this case v0.3.0 should consume less power. Another thing is that current implementation don't call (AFAIK) mtdev_idle function. Right now I suspect timers and scrolling, but you have to confirm.

hobarrera, 2017-08-17:
Same power consumption with v0.3.0. Is there any way to make sure if it's a hardware issue, rather than driver-related?

p2rkw, 2017-08-17:
Did you calibrated your powertop?

You may try to disable/enable your device with xinput. Eventually you can make this[1] function empty and recompile/reinstall mtrack as dummy driver, and then check power consumption.

[1] https://github.com/p2rkw/xf86-input-mtrack/blob/master/driver/mtrack.c#L500

p2rkw, 2017-08-20:
Powertop behaviour is strange because it reports strange numbers right after start:
```9.24 W     17,2 ms/s     3046,0       Interrupt      [7] INT33C3:00```
After few seconds report has changed to:
```Power est.              Usage       Events/s    Category       Description
  4.06 W      1,0%                      Device         Display backlight
  706 mW     59,8 ms/s      88,1        Process        /usr/lib/firefox-esr/firefox-esr
  677 mW      2,2 ms/s     220,8        Interrupt      [7] INT33C3:00
```

(interrupt comes from my touchpad)

closed 2018-01-21 by p2rkw