p2rkw/xf86-input-mtrack #46: touchpad speed increases when external display is plugged

anthrotype, 2017-10-09:
Hello,
thanks for this great touchpad drivers.
I'm using it on my Lenovo X1 Carbon 5th gen, running Debian Buster.

I noticed that when I plug an external display the cursor speed increases (almost doubles).
Is there anything I can do, besides of course running a shell script that calls xinput to adjust the sensitivity when the external display is plugged?

Thanks

p2rkw, 2017-10-10:
Haha sounds interesting :)
How do you set proffered values? Via xorg.conf or xinput in some custom script?
Chances are that values are re read from xorg.conf.
Does values reported by xinput --list-props differ after you connect external display?

anthrotype, 2017-10-10:
I set it via xorg.conf, here is my configuration file "/usr/share/X11/xorg.conf.d/50-mtrack.conf"

```
Section "InputClass"
        MatchIsTouchpad "on"
        Identifier      "Touchpads"
        Driver          "mtrack"
	Option		"AccelerationProfile" "2"
	Option		"AdaptiveDeceleration" "2"
	Option		"VelocityScale" "1.9"
        Option          "Sensitivity" "0.40"
        Option          "FingerHigh" "1"
        Option          "FingerLow" "1"
        Option          "IgnoreThumb" "true"
        Option          "ThumbRatio" "70"
        Option          "ThumbSize" "25"
        Option          "IgnorePalm" "true"
        Option          "TapButton1" "1"
        Option          "TapButton2" "3"
        Option          "TapButton3" "2"
        Option          "TapButton4" "0"
        Option          "ClickFinger1" "1"
        Option          "ClickFinger2" "3"
        Option          "ClickFinger3" "2"
        Option          "ButtonMoveEmulate" "false"
        Option          "ButtonIntegrated" "false"
        Option          "ScrollCoastDuration" "500"
        Option          "ScrollCoastEnableSpeed" ".3"
        Option          "ScrollUpButton" "5"
        Option          "ScrollDownButton" "4"
        Option          "ScrollLeftButton" "7"
        Option          "ScrollRightButton" "6"
        Option          "ScrollDistance" "200"
	Option		"RotateLeftButton" "9"
	Option		"RotateRightButton" "8"
EndSection
```

I've seen other similar reports online if you google for "touchpad speed increases with external display", etc. For example this:
https://superuser.com/questions/308966/touchpad-sensitivity-scales-to-virtual-screen-dimensions

the values from xinput list-props don't change after I plug in the external display.

As a workaround, I wrote a script to reduce the velocity scaling when I plug in the display to compensate for that. I found the value empirically, but maybe you can find a more elegant solution.

My internal display is 1920x1080, whereas my external one is 1920x1200, and it's positioned next to it on the right. I noticed the speed almost doubles, but I'm not sure.

Thanks for your help.

p2rkw, 2017-10-28:
Fix on master, please test.

anthrotype, 2017-11-10:
I pulled the latest  master and did the usual incantation (./configure && make && sudo make install) but the issue is still present, nothing seems to have changed :(

anthrotype, 2017-11-10:
is it something in my settings (pasted above) maybe?

p2rkw, 2018-03-12:
Are you still facing this issue?

WhyNotHugo, 2018-03-16:
I also have this same issue (always have, but have been postponing reporting it).

What can I provide to help debug it?

Thanks!

p2rkw, 2018-03-17:
Post output of xinput --list-props, I will try to reproduce again.

commit 01adea5d (2017-10-28): Fix #46: make movement speed screen size/resolution independent