p2rkw/xf86-input-mtrack #48: No left click on master

m4rcu5, 2017-10-22:
I just upgraded from v0.4.1 to master, upon which my single finger click changed to a right mouse click, and the two finger click seems to have changed to middle mouse.
This leaves me without a left mouse click option.

Hardware used is a Macbook Pro Retina 2015.

Below is my unchanged Xorg config for the mtrack module:
```
Section "InputClass"
  MatchIsTouchpad "on"
  Identifier "Touchpads"
  Driver "mtrack"

  # Adjusts the sensitivity (movement speed) of the touchpad. This is a real
  # number greater than or equal to zero. Default is 1. A value of 0 will
  # disable pointer movement.
  Option "Sensitivity" "0.55"

  # Defines the pressure at which a finger is detected as a touch. This is a
  # percentage represented as an integer. Default is 5.
  Option "FingerHigh" "5"

  # Defines the pressure at which a finger is detected as a release. This is a
  # percentage represented as an integer. Default is 5.
  Option "FingerLow" "2"

  # Whether or not to ignore touches that are determined to be thumbs. Boolean
  # value. Defaults to false.
  Option "IgnoreThumb" "false"

  # Whether or not to ignore touches that are determined to be palms. Boolean
  # value. Defaults to false.
  Option "IgnorePalm" "false"

  # Whether or not to disable the entire trackpad when a palm is touching.
  # Boolean value. Defaults to false.
  Option "DisableOnPalm" "true"

  # The size of an area at the bottom of the trackpad where new touches are
  # ignored (fingers traveling into this area from above will still be
  # tracked). This is represented as a percentage of the total trackpad height.
  # Defaults to 10.
  Option "BottomEdge" "25"

  # Whether or not the physical buttons are integrated with the trackpad. If
  # you have a one-piece trackpad like on newer MacBooks, this should be set to
  # true. Button emulation depends on this value being correct. Boolean value.
  # Defaults to true.
  Option "ButtonIntegrated" "true"

  # Whether or not to count the moving finger when emulating button clicks.
  # Useful to disable if you use two hands on trackpad. Boolean value. Defaults
  # to true.
  Option "ButtonMoveEmulate" "true"

  # Which button to emulate for one-finger tapping. Integer value. A value of 0
  # disables one-finger tapping. Defaults to 1.
  Option "TapButton1" "0"

  # Which button to emulate for two-finger tapping. Integer value. A value of 0
  # disables two-finger tapping. Defaults to 3.
  Option "TapButton2" "0"

  # Which button to emulate for three-finger tapping. Integer value. A value of
  # 0 disables three-finger tapping. Defaults to 2.
  Option "TapButton3" "0"

  # Which button to emulate for four-finger tapping. Integer value. A value of
  # 0 disables three-finger tapping. Defaults to 0.
  Option "TapButton4" "0"

  # When tapping, how much time to hold down the emulated button. Integer value
  # representing milliseconds. Defaults to 50.
  Option "ClickTime" "25"

  # For two finger scrolling. How far you must move your fingers before a
  # button click is triggered. Integer value. Defaults to 150.
  Option "ScrollDistance" "75"

  # For two finger scrolling. The button that is triggered by scrolling up.
  # Integer value. A value of 0 disables scrolling up. Defaults to 4.
  Option "ScrollUpButton" "5"

  # For two finger scrolling. The button that is triggered by scrolling down.
  # Integer value. A value of 0 disables scrolling down. Defaults to 5.
  Option "ScrollDownButton" "4"

  # For two finger scrolling. The button that is triggered by scrolling left.
  # Integer value. A value of 0 disables scrolling left. Defaults to 6.
  Option "ScrollLeftButton" "7"

  # For two finger scrolling. The button that is triggered by scrolling right. Integer value.
  # A value of 0 disables scrolling right. Defaults to 7.
  Option "ScrollRightButton" "6"

  # For three finger swiping. How far you must move your fingers before a
  # button click is triggered. Integer value. Defaults to 700.
  Option "SwipeDistance" "700"

  # For three finger swiping. The button that is triggered by swiping up.
  # Integer value. A value of 0 disables swiping up. Defaults to 8.
  Option "SwipeUpButton" "0"

  # For three finger swiping. The button that is triggered by swiping down.
  # Integer value. A value of 0 disables swiping down. Defaults to 9.
  Option "SwipeDownButton" "0"

  # For three finger swiping. The button that is triggered by swiping left. 
  # Integer value. A value of 0 disables swiping left. Defaults to 10.
  Option "SwipeLeftButton" "9"

  # For three finger swiping. The button that is triggered by swiping right.
  # Integer value. A value of 0 disables swiping right. Defaults to 11.
  Option "SwipeRightButton" "8"

  # Whether or not to enable tap-to-drag functionality. Boolean value. Defaults to true.
  Option "TapDragEnable" "false"

EndSection
```

p2rkw, 2017-10-23:
Please try set these configuration options first:
```
ClickFinger1 - Which button to emulate when one finger is touching the trackpad during a click. Integer value. A value of 0 disables one-touch emulation. Defaults to 3.
ClickFinger2 - Which button to emulate when two fingers are touching the trackpad during a click. Integer value. A value of 0 disabled one-touch emulation. Defaults to 2.
ClickFinger3 - Which button to emulate when three fingers are touching the trackpad during a click. Integer value. A value of 0 disabled one-touch emulation. Defaults to 0.
```
If it doesn't help paste output of `xinput --list-props <device_id>`

m4rcu5, 2017-10-28:
Hi @p2rkw,

Setting the "ClickFingers" fixed the issue indeed. I take ti the defaults have changed over time, causing my issue.

My device is a Macbook early 2015, If I understand the "IntegratedButton" feature, as a device with no separate mouse buttons, but you press anywhere on the mousepad to click, then yes.

p2rkw, 2017-10-28:
I think that commit https://github.com/p2rkw/xf86-input-mtrack/commit/ba7043100635372e74465369118470ca43ec13d8 just fixed buggy behavior, so ClickFinger* values started to be used as they should be.

closed 2017-10-28 by p2rkw