p2rkw/xf86-input-mtrack #50: Tap drag triggers double click

thatsmeta, 2017-11-11:
Hi!
I'm very happy with mtrack so far. However, tap drag does not seem to work correctly for me. When I tap once, then touch again and move my finger, apart from dragging, it also triggers a double click. So whenever I want to move a folder in my file explorer, it gets opened because of the double click.
Is there something wrong with my configuration? Any help is appreciated.

`
Device 'Synaptics TM3242-001':
	Device Enabled (141):	1
	Coordinate Transformation Matrix (143):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (275):	0
	Device Accel Constant Deceleration (276):	1.000000
	Device Accel Adaptive Deceleration (277):	1.000000
	Device Accel Velocity Scaling (278):	10.000000
	Trackpad Disable Input (279):	0
	Trackpad Sensitivity (280):	0.650000
	Trackpad Touch Pressure (281):	5, 5
	Trackpad Button Settings (282):	1, 1
	Trackpad Button Emulation Settings (283):	1, 1, 100
	Trackpad Button Emulation Values (284):	1, 3, 0
	Trackpad Tap Settings (285):	50, 120, 400
	Trackpad Tap Button Emulation (286):	1, 3, 2, 0
	Trackpad Thumb Detection (287):	0, 0
	Trackpad Thumb Size (288):	25, 70
	Trackpad Palm Detection (289):	1, 0
	Trackpad Palm Size (290):	40
	Trackpad Gesture Settings (291):	10, 100
	Trackpad Smooth Scroll (292):	1
	Trackpad Scroll Settings (293):	20, 40, 0
	Trackpad Scroll Buttons (294):	4, 5, 6, 7
	Trackpad Swipe Settings (295):	700, 300, 0
	Trackpad Swipe Buttons (296):	8, 9, 10, 11
	Trackpad Swipe4 Settings (297):	700, 300, 0
	Trackpad Swipe4 Buttons (298):	0, 0, 0, 0
	Trackpad Scroll Coasting (299):	0.100000, 500.000000
	Trackpad Edge Scroll Settings (300):	105, 20, 0, 4, 5, 6, 7
	Trackpad Edge Sizes (301):	0, 10, 0, 0
	Trackpad Scale Distance (302):	150
	Trackpad Scale Buttons (303):	12, 13
	Trackpad Rotate Distance (304):	150
	Trackpad Rotate Buttons (305):	14, 15
	Trackpad Hold1Move1 Stationary Settings (306):	20, 1
	Trackpad Hold1Move1 Settings (307):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (308):	1, 1, 1, 1
	Trackpad Drag Settings (309):	1, 350, 40, 200
	Trackpad Axis Inversion (310):	0, 0
`

`
Section "InputClass"
MatchIsTouchpad "on"
Identifier "Touchpads"
Driver "mtrack"
Option "Sensitivity" "0.65"
Option "ClickFinger1" "1"
Option "ClickFinger2" "3"
Option "ClickFinger3" "0"
Option "ButtonZonesEnable" "true"
Option "ScrollDistance" "20"
Option "ScrollCoastDuration" "500"
Option "ScrollClickTime" "40"
Option "IgnorePalm" "true"
Option "TapDragEnable" "true
EndSection
`

p2rkw, 2017-11-18:
Was it always like that? Did this changed with some update?

I suggest you to use "Hold and move" feature: I described it described in readme.

Anyways I will what can be done here, but to implement double tap as you one, every tap will have to be postponed by some delay.

HactarCE, 2019-01-17:
> Anyways I will what can be done here, but to implement double tap as you one, every tap will have to be postponed by some delay.

This is exactly what most touchpad drivers do. If it's user-configurable, I see no problem with this. If the user needs responsive clicking, they can either use "hard" click or configure a small/zero delay.

xavierhd, 2019-01-18:
From what I've seen, it has always been like that.  
The click event should only be sent if the finger is lifted fast enough after the second tap.