p2rkw/xf86-input-mtrack #51: Circular scrolling

altayhunter, 2017-11-30:
One of my favorite features of the xf86-input-synaptics driver is [CircularScrolling](https://www.x.org/archive//X11R7.5/doc/man/man4/synaptics.4.html), which is a feature similar to the chiral scrolling option in some of Synaptics' drivers for Windows.
In particular, there is an excellent patent-free implementation of it [available as a patch](https://gist.github.com/galundin/736d7331a626a0deda13) over in [this GitHub project](https://github.com/galundin/circular-scrolling-improved). I would be thrilled if this feature were incorporated into the mtrack driver.

p2rkw, 2017-12-05:
Interesting. 
It could be done in mtrack and it would be really unique feature, but I see few problems here:
 - as I checked on demo page http://galundin.github.io/circular-scrolling-improved/ fast back and forth movement (even straight) causes scrolling,
 - there's very little to no demand for that feature - I bet very few will discover it, and even less will be interested enough to try that,
 - even if implemented that feature will need to seat long time in development branch,
 - patents.

I recently implemented edge scrolling, did you try it? You can even set sensitivity for fast scrolling: https://github.com/p2rkw/xf86-input-mtrack#EdgeScrollSensitivity

altayhunter, 2017-12-05:
Thanks for your response.
- Fast back and forth resulting in scrolling does seem to be a limitation of the driver, although I think it could be fixed with more aggressive filtering by discarding delta angles close to pi. As an every-day user of this driver patch it's not something I was even aware of.
- This is probably true. There are definitely [still](https://answers.microsoft.com/en-us/windows/forum/windows_10-hardware/lenovo-flex-4-synaptics-touchpad-chiral-scrolling/43b27a22-7932-459b-a472-affc9157d4f7?auth=1) [people](https://www.reddit.com/r/Windows10/comments/6ny9nr/does_windows_precision_drivers_have_chiral/) interested in this feature, but it's admittedly a small set of holdouts who likely have no knowledge of this driver.
- I expect any new feature/gesture would spend a good deal of time in dev.
- This implementation is specifically designed to avoid violating [US7446754](https://www.google.com/patents/US7446754), but I understand if the risk isn't worth it regardless. To be honest I find it quite frustrating that Synaptics holds this patent but no longer implements it in their official drivers.

I'm personally not a big fan of edge scrolling, and I actually really like the idea of having circular scrolling being triggered by two-finger touch rather than from an edge. This is admittedly a niche enough idea that I may be the only one interested in it.
My problem with every other implementation of scrolling is having to lift my fingers off the touchpad frequently to scroll great distances. This can be mitigated by increasing acceleration or enabling momentum, but I find that these options reduce precision.

Anyway, thanks again for your response.