p2rkw/xf86-input-mtrack #52: synaptics 3/4 finger gestures

ethician, 2017-12-15:
First, thanks for the hard work in keeping this project up!

Would you consider adding handling of Synaptics' poor reporting of multitouch gestures?

As evidenced by `evtest`, Synaptics unfortunately reports third, and higher, tap as a sequence of `BTN_TOUCH` (or `BTN_FINGER`) followed by `BTN_TOOL_TRIPLETAP` (or `BTN_TOOL_QUADTAP`, etc...).

```
Event: time 1513311070.511217, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 0
Event: time 1513311070.511217, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 1892
Event: time 1513311070.511217, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 3141
Event: time 1513311070.511217, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 3851
Event: time 1513311070.511217, type 3 (EV_ABS), code 47 (ABS_MT_SLOT), value 1
Event: time 1513311070.511217, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 1893
Event: time 1513311070.511217, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 4276
Event: time 1513311070.511217, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1984
Event: time 1513311070.511217, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 48
Event: time 1513311070.511217, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1513311070.511217, type 3 (EV_ABS), code 0 (ABS_X), value 3141
Event: time 1513311070.511217, type 3 (EV_ABS), code 1 (ABS_Y), value 3851
Event: time 1513311070.511217, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 56
Event: time 1513311070.511217, type 1 (EV_KEY), code 334 (BTN_TOOL_TRIPLETAP), value 1
```

`libinput` does handle it correctly (relevant code in `libinput/src/evdev-mt-touchpad.c` around `tp_fake_finger...`) but lacks mtrack's other features, notably superb kinetic scrolling.

Thanks for the consideration!

p2rkw, 2017-12-16:
Your device can track 2 fingers at once. How does synaptics report movement after that 3 finger tap (BTN_TOOL_TRIPLETAP=1) ? Does it generate events for slot 0, 1 or for both?

ethician, 2017-12-18:
Well, it can and it can do more than that. The log I posted from my own device shows that it uses both slots 0 and 1, but it also reports the other touches using `BTN_TOUCH` (sometimes `BTN_FINGER`) in conjunction with the number-of-fingers event like `BTN_TOOL_TRIPLETAP`.

The code in `libinput` I pointed to does handle these compound events and correctly reports 3 and 4 finger gestures via `libinput-debug-events`.

If this is something worth considering, let me know what additional info I can provide you with.