p2rkw/xf86-input-mtrack #54: Tap Drag Lock Timeout

CapSel, 2018-01-19:
I can't find a way to enable behaviour from libinput driver - Tap Drag Lock Timeout. Is it possible with this driver? If not could you add support for it?

Tap Drag Lock Timout is time that TapDrag waits for your finger to re-touch touchpad during tap-to-drag so that you can things for loooong distances, selecting really long text.

The only similar behaviour I found is emulating dragging with keyboard cursor keys.
My current settings:
 xinput list-props 11
Device 'ELAN1200:00 04F3:3022 Touchpad':
        Device Enabled (141):   1
        Coordinate Transformation Matrix (143): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (276):     0
        Device Accel Constant Deceleration (277):       1.000000
        Device Accel Adaptive Deceleration (278):       1.000000
        Device Accel Velocity Scaling (279):    10.000000
        Trackpad Disable Input (280):   0
        Trackpad Sensitivity (281):     0.600000
        Trackpad Touch Pressure (282):  5, 5
        Trackpad Button Settings (283): 1, 1
        Trackpad Button Emulation Settings (284):       0, 1, 100
        Trackpad Button Emulation Values (285): 3, 2, 0
        Trackpad Tap Settings (286):    5, 120, 400
        Trackpad Tap Button Emulation (287):    1, 3, 2, 0
        Trackpad Thumb Detection (288): 0, 0
        Trackpad Thumb Size (289):      25, 70
        Trackpad Palm Detection (290):  0, 0
        Trackpad Palm Size (291):       40
        Trackpad Gesture Settings (292):        10, 100
        Trackpad Smooth Scroll (293):   1
        Trackpad Scroll Settings (294): 150, 20, 0
        Trackpad Scroll Buttons (295):  4, 5, 6, 7
        Trackpad Swipe Settings (296):  700, 300, 0
        Trackpad Swipe Buttons (297):   8, 9, 10, 11
        Trackpad Swipe4 Settings (298): 700, 300, 0
        Trackpad Swipe4 Buttons (299):  0, 0, 0, 0
        Trackpad Scroll Coasting (300): 0.400000, 400.000000
        Trackpad Scale Distance (301):  150
        Trackpad Scale Buttons (302):   12, 13
        Trackpad Rotate Distance (303): 150
        Trackpad Rotate Buttons (304):  14, 15
        Trackpad Hold1Move1 Stationary Settings (305):  20, 1
        Trackpad Hold1Move1 Settings (306):     1, 0, 1000
        Trackpad Hold1Move1 Buttons (307):      1, 1, 1, 1
        Trackpad Drag Settings (308):   1, 350, 40, 200
        Trackpad Axis Inversion (309):  0, 0
        Trackpad Edge Size (310):       10

Gentoo.

p2rkw, 2018-01-21:
Please test this branch: https://github.com/p2rkw/xf86-input-mtrack/tree/tap_drag_tiemout

I hardcoded drag lock time to 500 ms, but it could potentially be infinite, which means additional tap will be needed to break drag mode.

CapSel, 2018-01-21:
Works excellent!
libinput uses hardcoded 500ms.
Is there a way to disable keyboard cursor arrows based window movement?

p2rkw, 2018-01-21:
New commits, please test, and let me know about issues before I push to master.

CapSel, 2018-01-22:
Now: Cursor movement is disabled but the cross arrows are visible and window (in KDE) is semi-transparent = waiting for drag finish.

Now and previous version of this branch:
There is also a problem with clicking before drag - no matter how fast I try to tap to drag there is always double click and drag. Is it possible to wait with "release" of the button for some time (maybe 150ms) to see if user decides to tap again or drag?

p2rkw, 2018-01-23:
> Now: Cursor movement is disabled but the cross arrows are visible and window (in KDE) is semi-transparent = waiting for drag finish.

I don't fully understand: does it mean my latest changes disable your cursor movement?

> Now and previous version of this branch:
> There is also a problem with clicking before drag - no matter how fast I try to tap to drag there is always double click and drag. Is it possible to wait with "release" of the button for some time (maybe 150ms) to see if user decides to tap again or drag?

I think I can tweak code to support that, but it will require some delay for tap clicks. See #50

CapSel, 2018-01-23:
When I start drag mouse cursor is as a 4-way-arrows, window is semi-transparent.

If I finish drag by waiting 500ms:
- with previous version: mouse cursor stays 4-way-arrows, window is still semi-transparent, I can use arrow keys from keyboard to move window. Tapping or enter finishes drag.
- with current version: mouse cursor stays 4-way-arrows, window is still semi-transparent, I can't use arrow keys from keyboard to move window. Tapping finishes drag.