p2rkw/xf86-input-mtrack #57: cannot configure device error

jeremyj, 2018-02-06:
Hello,

When loading mtrack to control my touchpad I'm getting the following errors in Xorg.0.log:

```
[07:59:20] jeremy@revo:~$ grep mtrack /var/log/Xorg.0.log 
[  3213.767] (II) LoadModule: "mtrack"
[  3213.767] (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
[  3213.768] (II) Module mtrack: vendor="X.Org Foundation"
[  3213.768] (II) Using input driver 'mtrack' for 'DLL075B:01 06CB:76AF Touchpad'
[  3213.770] (II) mtrack[1] driver/mtrack.c:564: (II) device control: init
[  3213.770] (II) mtrack[1] src/capabilities.c:183: (II) Capabilities: 
[  3213.770] (II) mtrack[...]: devname: DLL075B:01 06CB:76AF Touchpad
[  3213.770] (II) mtrack[...]: devid: 6cb 76af 100
[  3213.770] (II) mtrack[...]: ABS_MT_POSITION_X: min: 0 max: 1216
[  3213.770] (II) mtrack[...]: ABS_MT_POSITION_Y: min: 0 max: 680
[  3213.770] (II) mtrack[...]: ABS_MT_TRACKING_ID: min: 0 max: 65535
[  3213.770] (II) mtrack[...]: ABS_MT_SLOT: min: 0 max: 4
[  3213.770] (II) mtrack[...]: enabled caps: left mtdata slot
[  3213.792] (II) mtrack[2] driver/mtrack.c:567: (II) device control: on
[  3213.793] (WW) mtrack[1] src/mconfig.c:179: (WW) Touchpad has minimal capabilities. Some features will be unavailable.
[  3213.794] (II) Using input driver 'mtrack' for 'DLL075B:01 06CB:76AF Touchpad'
[  3213.796] (II) mtrack[3] driver/mtrack.c:564: (II) device control: init
[  3213.796] (EE) mtrack[4] driver/mtrack.c:152: (EE) cannot configure device DLL075B:01 06CB:76AF Touchpad
[  3213.796] (II) UnloadModule: "mtrack"
```
Can someone tell me if this a problem with the device that is incompatible?
Has anyone successfully used mtrack with a Dell DLL075B:01 06CB:76AF Touchpad on a Dell XPS 13 (9360)?

Thanks

p2rkw, 2018-02-06:
Hi,
I'm using mtrack with Dell XPS 13 9333, works fine, please attach whole Xorg.0.log file.

jeremyj, 2018-02-06:
https://pastebin.com/sLvF3aqF

using minimal configuration

```
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier      "Touchpads"
    Driver          "mtrack"
EndSection
```

jeremyj, 2018-02-11:
Any news on this issue? Thanks

p2rkw, 2018-02-11:
Hi,
grom very quick glance at your log I can tell you have two instances of mtrack fighting for one device:
```
[ 10481.759] (II) config/udev: Adding input device DLL075B:01 06CB:76AF Touchpad (/dev/input/event17)
[ 10481.762] (II) XINPUT: Adding extended input device "DLL075B:01 06CB:76AF Touchpad" (type: TOUCHPAD, id 12)
[ 10481.762] (II) mtrack[1] driver/mtrack.c:564: (II) device control: init

(later in log)

[ 10481.782] (II) config/udev: Adding input device DLL075B:01 06CB:76AF Touchpad (/dev/input/mouse1)
[ 10481.782] (II) XINPUT: Adding extended input device "DLL075B:01 06CB:76AF Touchpad" (type: TOUCHPAD, id 13)
[ 10481.783] (II) mtrack[3] driver/mtrack.c:564: (II) device control: init
[ 10481.783] (**) Option "Device" "/dev/input/mouse1"
[ 10481.783] (EE) mtrack[4] driver/mtrack.c:152: (EE) cannot configure device DLL075B:01 06CB:76AF Touchpad
[ 10481.783] (EE) Couldn't init device "DLL075B:01 06CB:76AF Touchpad"
[ 10481.783] (II) UnloadModule: "mtrack"
```
Something strange is going on here.
Same apply to your integrated webcam - does it work?
Please paste output of `xinput --list`

p2rkw, 2018-02-11:
So with some googling I found potential solution, add:
```
    MatchDevicePath "/dev/input/event*"
```
to your conf file.

jeremyj, 2018-02-11:
adding `MatchDevicePath "/dev/input/event*"` to my conf file seems to work. Now I'm having some trouble configuring the dirver to meet my need. More on that later :-)
Thanks!

closed 2018-02-12 by p2rkw