p2rkw/xf86-input-mtrack #59: Two finger scroll gets speedup if another finger is accidentally added

svenper, 2018-02-19:
When scrolling with two fingers, and adding a third finger for a split second and then continuing scrolling, the speed is increased what seems to be 10-fold or even 100-fold. After the speedup has occurred, while still scrolling, and tapping with a third finger another time, the speed goes back to normal.

Version 0.5.0
Linux 4.14.19 (musl libc)
MacBookPro 12,1

p2rkw, 2018-02-24:
Scrolling with third finger on touchpad will be recognized as 3 finger drag. Thus mtrack will send configured button. Actual action taken depend on your setup. On my machine it was toggling fullscreen.

Before I start investigating this, please you have to confirm it's caused by mtrack.

How to check that:
 - run `xinput --test <dev id>`
 - reproduce your issue with xinput running
 - check in output which button was sent
 - send that button with xdotool or something else

If speedup occurs with xdotool, it's not caused with mtrack, otherwise post your configuration (`xinput --list-props`).

svenper, 2018-02-25:
I should also say that the added third finger can have two outcomes, depending mostly on how long the tap is and if the third finger also moves, I think.

* With a long *deliberate movement* I can get the same behaviour you describe, with both events for two finger and three finger scroll (e.g. 4 and 9).
* With a short *accidental tap* I get the speedup.

knedlsepp, 2018-02-27:
I'm also seeing this behavior.

p2rkw, 2018-02-27:
Ok, I need your output of `xinput --list-props <dev ID>`.

knedlsepp, 2018-02-27:
```
Device 'bcm5974':                                                                                                                                            
        Device Enabled (138):   1                                                                                                                            
        Coordinate Transformation Matrix (140): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000                     
        Device Accel Profile (275):     2                                                                                                                    
        Device Accel Constant Deceleration (276):       2.100000                                                                                             
        Device Accel Adaptive Deceleration (277):       1.000000                                                                                             
        Device Accel Velocity Scaling (278):    1.000000                                                                                                     
        Trackpad Disable Input (279):   0                                                                                                                    
        Trackpad Sensitivity (280):     0.240000                                                                                                             
        Trackpad Touch Pressure (281):  5, 1                                                                                                                 
        Trackpad Button Settings (282): 1, 1                                                                                                                 
        Trackpad Button Emulation Settings (283):       0, 1, 10                                                                                             
        Trackpad Button Emulation Values (284): 1, 3, 2
        Trackpad Tap Settings (285):    50, 150, 400
        Trackpad Tap Button Emulation (286):    1, 3, 2, 0
        Trackpad Thumb Detection (287): 1, 0
        Trackpad Thumb Size (288):      22, 90
        Trackpad Palm Detection (289):  1, 1
        Trackpad Palm Size (290):       40
        Trackpad Gesture Settings (291):        10, 0
        Trackpad Smooth Scroll (292):   1
        Trackpad Scroll Settings (293): 180, 20, 0
        Trackpad Scroll Buttons (294):  4, 5, 6, 7
        Trackpad Swipe Settings (295):  1000, 300, 0
        Trackpad Swipe Buttons (296):   11, 10, 8, 9
        Trackpad Swipe4 Settings (297): 700, 300, 0
        Trackpad Swipe4 Buttons (298):  0, 0, 0, 0
        Trackpad Scroll Coasting (299): 0.050000, 1000.000000
        Trackpad Edge Scroll Settings (300):    105, 20, 0, 0, 0, 0, 0
        Trackpad Edge Sizes (301):      0, 0, 0, 15
        Trackpad Scale Distance (302):  150
        Trackpad Scale Buttons (303):   12, 13
        Trackpad Rotate Distance (304): 150
        Trackpad Rotate Buttons (305):  14, 15
        Trackpad Hold1Move1 Stationary Settings (306):  20, 0
        Trackpad Hold1Move1 Settings (307):     1, 0, 1000
        Trackpad Hold1Move1 Buttons (308):      1, 1, 1, 1
        Trackpad Drag Settings (309):   0, 1, 40, 200, 500
        Trackpad Axis Inversion (310):  0, 0
```

svenper, 2018-02-27:
```
Device 'bcm5974':
	Device Enabled (141):	1
	Coordinate Transformation Matrix (143):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (265):	0
	Device Accel Constant Deceleration (266):	1.000000
	Device Accel Adaptive Deceleration (267):	1.000000
	Device Accel Velocity Scaling (268):	10.000000
	Trackpad Disable Input (298):	0
	Trackpad Sensitivity (299):	0.125000
	Trackpad Touch Pressure (300):	5, 5
	Trackpad Button Settings (301):	1, 1
	Trackpad Button Emulation Settings (302):	0, 1, 100
	Trackpad Button Emulation Values (303):	1, 3, 2
	Trackpad Tap Settings (304):	50, 120, 400
	Trackpad Tap Button Emulation (305):	1, 3, 2, 0
	Trackpad Thumb Detection (306):	1, 0
	Trackpad Thumb Size (307):	25, 70
	Trackpad Palm Detection (308):	1, 0
	Trackpad Palm Size (309):	40
	Trackpad Gesture Settings (310):	10, 50
	Trackpad Smooth Scroll (311):	1
	Trackpad Scroll Settings (312):	150, 20, 0
	Trackpad Scroll Buttons (313):	5, 4, 7, 6
	Trackpad Swipe Settings (314):	700, 300, 0
	Trackpad Swipe Buttons (315):	8, 9, 10, 11
	Trackpad Swipe4 Settings (316):	700, 300, 0
	Trackpad Swipe4 Buttons (317):	0, 0, 0, 0
	Trackpad Scroll Coasting (318):	0.100000, 200.000000
	Trackpad Edge Scroll Settings (319):	105, 20, 0, 4, 5, 6, 7
	Trackpad Edge Sizes (320):	0, 10, 0, 0
	Trackpad Scale Distance (321):	150
	Trackpad Scale Buttons (322):	12, 13
	Trackpad Rotate Distance (323):	150
	Trackpad Rotate Buttons (324):	14, 15
	Trackpad Hold1Move1 Stationary Settings (325):	20, 1
	Trackpad Hold1Move1 Settings (326):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (327):	1, 1, 1, 1
	Trackpad Drag Settings (328):	1, 350, 40, 200, 500
	Trackpad Axis Inversion (329):	0, 0
```

p2rkw, 2018-02-28:
Thanks.

I suspect smooth scroll or coasting to have some impact on this. You can try to disable these and see does it change something or not.

svenper, 2018-02-28:
I set both `ScrollCoastDuration` and `ScrollCoastEnableSpeed` to `0` and cannot reproduce the bug anymore. When scrolling, tapping *or* holding the third finger now instead stops both `button` and `motion` events. Keeping the single finger pressed and continuing to scroll after this can trigger three-finger scrolls, but very unreliaby.