p2rkw/xf86-input-mtrack #60: Trackpad speed way too fast after update

0x783czar, 2018-02-24:
After updating to the latest version, suddenly my trackpad speed is out of control.  It's incredibly fast and it seems the smallest increment my pointer can move is about 10 pixels at a time.

- **Version: xf86-input-mtrack-git 0.5.0.r2.g0185630-1**
- **OS: Arch Linux**
- **Kernel: 4.15.5-1-ARCH**
- **Hardware: Macbook Pro 8,2**

```
Device 'bcm5974':
	Device Enabled (141):	1
	Coordinate Transformation Matrix (143):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (276):	0
	Device Accel Constant Deceleration (277):	1.000000
	Device Accel Adaptive Deceleration (278):	1.000000
	Device Accel Velocity Scaling (279):	10.000000
	Trackpad Disable Input (280):	0
	Trackpad Sensitivity (281):	1.000000
	Trackpad Touch Pressure (282):	5, 5
	Trackpad Button Settings (283):	1, 1
	Trackpad Button Emulation Settings (284):	0, 1, 100
	Trackpad Button Emulation Values (285):	1, 2, 3
	Trackpad Tap Settings (286):	50, 120, 400
	Trackpad Tap Button Emulation (287):	1, 3, 2, 0
	Trackpad Thumb Detection (288):	0, 0
	Trackpad Thumb Size (289):	50, 70
	Trackpad Palm Detection (290):	0, 0
	Trackpad Palm Size (291):	40
	Trackpad Gesture Settings (292):	10, 100
	Trackpad Smooth Scroll (293):	1
	Trackpad Scroll Settings (294):	100, 20, 0
	Trackpad Scroll Buttons (295):	5, 4, 7, 6
	Trackpad Swipe Settings (296):	700, 300, 0
	Trackpad Swipe Buttons (297):	8, 9, 10, 11
	Trackpad Swipe4 Settings (298):	700, 300, 0
	Trackpad Swipe4 Buttons (299):	0, 0, 0, 0
	Trackpad Scroll Coasting (300):	0.100000, 200.000000
	Trackpad Edge Scroll Settings (301):	105, 20, 0, 4, 5, 6, 7
	Trackpad Edge Sizes (302):	0, 10, 0, 0
	Trackpad Scale Distance (303):	150
	Trackpad Scale Buttons (304):	12, 13
	Trackpad Rotate Distance (305):	150
	Trackpad Rotate Buttons (306):	14, 15
	Trackpad Hold1Move1 Stationary Settings (307):	20, 1
	Trackpad Hold1Move1 Settings (308):	1, 0, 1000
	Trackpad Hold1Move1 Buttons (309):	1, 1, 1, 1
	Trackpad Drag Settings (310):	1, 350, 40, 200, 500
	Trackpad Axis Inversion (311):	0, 0
```

```
Section "InputClass"
	Identifier "Multitouch Touchpad"
	Driver "mtrack"
	#MatchDevicePath "/dev/input/event*"
	MatchProduct "bcm5974"
	MatchIsTouchpad "true"
	option "ScrollUpButton" "5"
	option "ScrollDownButton" "4"
	option "ScrollLeftButton" "7"
	option "ScrollRightButton" "6"
	option "Thumbsize" "50"
	option "ScrollDistance" "100"
	option "Sensitivity" "1"
EndSection
```

p2rkw, 2018-02-26:
Interesting,
please check speed with smaller acceleration:
 `xinput --set-prop <dev> 'Device Accel Velocity Scaling' 1`

p2rkw, 2018-02-26:
That could be caused by https://github.com/p2rkw/xf86-input-mtrack/commit/01adea5d9d690b346090da116cc79677c96573e5.
What is resolution of your virtual desktop?

closed 2018-03-15 by p2rkw

AkitakiKou, 2018-06-08:
I can confirm this issue, on my MacBook Pro 12,1. The pointer moves so fast even with `Option "VelocityScale" "1.0"` properly set in Xorg.conf.

- OS: elementary OS 0.4.1
- Kernel: 4.13.0-43-generic
- Version: xf86-input-mtrack-git 0.5.0.r2.