p2rkw/xf86-input-mtrack #61: Hardware clicks using the thumb are ignored with IgnoreThumb on

knedlsepp, 2018-03-02:
While IgnoreThumb does what it says, the behavior of completely ignoring hardware clicks should be relaxed a little.

p2rkw, 2018-03-03:
While easy to change, it's very questionable, because it was made on purpose here: https://github.com/p2rkw/xf86-input-mtrack/commit/f92968de16f8f935c7be250f3c741aa2b261c66a

Personally I don't use physical button so I don't have strong stance on that, but I rarely change established behavior.

Just comment/remove this https://github.com/p2rkw/xf86-input-mtrack/blob/00674306aeea309a954285a8e0509c0755d6b20b/src/gestures.c#L338 line, and it should be fine.

knedlsepp, 2018-03-07:
Thanks for the suggestion. I tried it, but this didn't quite work out the way I expected it to.

p2rkw, 2018-03-08:
You can try it with changes from PR 62: that will allow hardware clicks on edges.