p2rkw/xf86-input-mtrack #63: mtrack unable to configure device

majuscule, 2018-03-08:
mtrack seems to be unable to load my trackpad. I'm suspicious that this may have something to do with udev/xorg loading multiple drivers for the single device (`input21/event18` vs `input21/mouse`), but I'm not sure. I've attached the log output from Xorg. Is there any way to get more debug info out of mtrack directly? For more fun, unplugging and replugging the trackpad in after this will crash Xorg entirely (`Server terminated with error (1)`). Happy to provide any other information I can.

Thanks!

```
[   769.446] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event18)
[   769.446] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
[   769.446] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Trackpads"
[   769.446] (II) LoadModule: "mtrack"
[   769.447] (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
[   769.447] (II) Module mtrack: vendor="X.Org Foundation"
[   769.447]    compiled for 1.19.6, module version = 0.2.0
[   769.447]    Module class: X.Org XInput Driver
[   769.447]    ABI class: X.Org XInput driver, version 24.1
[   769.447] (II) Using input driver 'mtrack' for 'Apple Inc. Magic Trackpad 2'
[   769.447] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   769.447] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   769.447] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.1/0003:05AC:0265.0012/input/input21/event18"
[   769.447] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 12)
[   769.447] (II) device control: init
[   769.447] (**) Option "Device" "/dev/input/event18"
[   769.499] (II) mtrack: devname: Apple Inc. Magic Trackpad 2
[   769.499] (II) mtrack: devid: 5ac 265 110
[   769.499] (II) mtrack: caps: left middle right
[   769.534] (**) Apple Inc. Magic Trackpad 2: (accel) keeping acceleration scheme 1
[   769.534] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration profile 0
[   769.534] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[   769.534] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[   769.534] (II) device control: on
[   769.586] (WW) Touchpad has minimal capabilities. Some features will be unavailable.
[   769.587] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse1)
[   769.587] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Trackpads"
[   769.587] (II) Using input driver 'mtrack' for 'Apple Inc. Magic Trackpad 2'
[   769.587] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   769.587] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   769.587] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.1/0003:05AC:0265.0012/input/input21/mouse1"
[   769.587] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 13)
[   769.587] (II) device control: init
[   769.588] (**) Option "Device" "/dev/input/mouse1"
[   769.588] (EE) mtrack: cannot configure device
[   769.588] (EE) Couldn't init device "Apple Inc. Magic Trackpad 2"
[   769.588] (II) UnloadModule: "mtrack"
```

p2rkw, 2018-03-10:
https://github.com/p2rkw/xf86-input-mtrack/issues/57#issuecomment-364782810

majuscule, 2018-03-10:
Hi @p2rkw, thank you! That seems to have solved my issue, however the trackpad is unresponsive. I'm guessing it is related to this issue: https://github.com/BlueDragonX/xf86-input-mtrack/issues/116. Is there any way to get additional debugging information, plans for testing with this device, or a way for me to contribute?

majuscule, 2018-03-11:
Unfortunately, that had no effect.

p2rkw, 2018-03-11:
So, do you forced mtrack to use /dev/input/event*? If so you can try switching to /dev/input/mouse*. If that will not help, post updated Xorg.log

majuscule, 2018-03-11:
Yes. With `/dev/input/event*` I see the warning listed in the linked issue (`[ 87676.982] (WW) Touchpad has minimal capabilities. Some features will be unavailable.`). With `/dev/input/mouse*`, I see `[ 87612.281] (EE) Couldn't init device "Apple Inc. Magic Trackpad 2`.

Here is the Xorg.log using event*:

```
[ 87676.850] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event19)
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Touchpads"
[ 87676.850] (II) LoadModule: "mtrack"
[ 87676.850] (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
[ 87676.850] (II) Module mtrack: vendor="X.Org Foundation"
[ 87676.850]    compiled for 1.19.6, module version = 0.2.0
[ 87676.850]    Module class: X.Org XInput Driver
[ 87676.850]    ABI class: X.Org XInput driver, version 24.1
[ 87676.850] (II) Using input driver 'mtrack' for 'Apple Inc. Magic Trackpad 2'
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: always reports core events
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: always reports core events
[ 87676.851] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:05AC:0265.0013/input/input23/event19"
[ 87676.851] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 12)
[ 87676.851] (II) device control: init
[ 87676.851] (**) Option "Device" "/dev/input/event19"
[ 87676.903] (II) mtrack: devname: Apple Inc. Magic Trackpad 2
[ 87676.903] (II) mtrack: devid: 5ac 265 110
[ 87676.903] (II) mtrack: caps: left middle right
[ 87676.930] (**) Apple Inc. Magic Trackpad 2: (accel) keeping acceleration scheme 1
[ 87676.930] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration profile 0
[ 87676.930] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[ 87676.930] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[ 87676.930] (II) device control: on
[ 87676.982] (WW) Touchpad has minimal capabilities. Some features will be unavailable.
[ 87676.983] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse1)
[ 87676.983] (II) No input driver specified, ignoring this device.
[ 87676.850] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event19)
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Touchpads"
[ 87676.850] (II) LoadModule: "mtrack"
[ 87676.850] (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
[ 87676.850] (II) Module mtrack: vendor="X.Org Foundation"
[ 87676.850]    compiled for 1.19.6, module version = 0.2.0
[ 87676.850]    Module class: X.Org XInput Driver
[ 87676.850]    ABI class: X.Org XInput driver, version 24.1
[ 87676.850] (II) Using input driver 'mtrack' for 'Apple Inc. Magic Trackpad 2'
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: always reports core events
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: always reports core events
[ 87676.851] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:05AC:0265.0013/input/input23/event19"
[ 87676.851] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 12)
[ 87676.851] (II) device control: init
[ 87676.851] (**) Option "Device" "/dev/input/event19"
[ 87676.903] (II) mtrack: devname: Apple Inc. Magic Trackpad 2
[ 87676.903] (II) mtrack: devid: 5ac 265 110
[ 87676.903] (II) mtrack: caps: left middle right
[ 87676.930] (**) Apple Inc. Magic Trackpad 2: (accel) keeping acceleration scheme 1
[ 87676.930] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration profile 0
[ 87676.930] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[ 87676.930] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[ 87676.930] (II) device control: on
[ 87676.982] (WW) Touchpad has minimal capabilities. Some features will be unavailable.
[ 87676.983] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse1)
[ 87676.983] (II) No input driver specified, ignoring this device.
[ 87676.850] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event19)
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Touchpads"
[ 87676.850] (II) LoadModule: "mtrack"
[ 87676.850] (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
[ 87676.850] (II) Module mtrack: vendor="X.Org Foundation"
[ 87676.850]    compiled for 1.19.6, module version = 0.2.0
[ 87676.850]    Module class: X.Org XInput Driver
[ 87676.850]    ABI class: X.Org XInput driver, version 24.1
[ 87676.850] (II) Using input driver 'mtrack' for 'Apple Inc. Magic Trackpad 2'
[ 87676.850] (**) Apple Inc. Magic Trackpad 2: always reports core events
[... 47 lines cut ...]
        Trackpad Tap Button Emulation (317):    1, 3, 2, 0
        Trackpad Thumb Detection (318): 0, 0
        Trackpad Thumb Size (319):      25, 70
        Trackpad Palm Detection (320):  0, 0
        Trackpad Palm Size (321):       40
        Trackpad Gesture Settings (322):        10, 100
        Trackpad Smooth Scroll (323):   1
        Trackpad Scroll Settings (324): 150, 20, 0
        Trackpad Scroll Buttons (325):  4, 5, 6, 7
[... cut ...]
        Trackpad Scale Distance (331):  150
        Trackpad Scale Buttons (332):   12, 13
        Trackpad Rotate Distance (333): 150
        Trackpad Rotate Buttons (334):  14, 15
        Trackpad Hold1Move1 Stationary Settings (335):  20, 1
        Trackpad Hold1Move1 Settings (336):     1, 0, 1000
        Trackpad Hold1Move1 Buttons (337):      1, 1, 1, 1
        Trackpad Drag Settings (338):   1, 350, 40, 200
        Trackpad Axis Inversion (339):  0, 0
        Trackpad Edge Size (340):       10
```

majuscule, 2018-03-11:
It may be worth mentioning that using `MatchIsTouchpad "on"` seems to prevent the `InputClass` from matching, which is why I switched to `MatchProduct`.

p2rkw, 2018-03-11:
It seems this device reports button events via event* and movement via mouse*. 
Please paste log when you try to load only with mouse*.

majuscule, 2018-03-11:
`Xorg.0.log` with `mouse*`:

```
[ 95412.114] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event19)
[ 95412.114] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
[ 95412.114] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[ 95412.114] (**) Apple Inc. Magic Trackpad 2: always reports core events
[ 95412.114] (**) Option "Device" "/dev/input/event19"
[ 95412.114] (**) Option "_source" "server/udev"
[ 95412.166] (II) event19 - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[ 95412.166] (II) event19 - Apple Inc. Magic Trackpad 2: device is a pointer
[ 95412.166] (II) event19 - Apple Inc. Magic Trackpad 2: device removed
[ 95412.194] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:05AC:0265.0013/input/input23/event19"
[ 95412.194] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: MOUSE, id 12)
[ 95412.194] (**) Option "AccelerationScheme" "none"
[ 95412.194] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[ 95412.194] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[ 95412.194] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[ 95412.246] (II) event19 - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[ 95412.246] (II) event19 - Apple Inc. Magic Trackpad 2: device is a pointer
[ 95412.248] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse1)
[ 95412.248] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Touchpads"
[ 95412.248] (II) LoadModule: "mtrack"
[ 95412.248] (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
[ 95412.248] (II) Module mtrack: vendor="X.Org Foundation"
[ 95412.248]    compiled for 1.19.6, module version = 0.2.0
[ 95412.248]    Module class: X.Org XInput Driver
[ 95412.248]    ABI class: X.Org XInput driver, version 24.1
[ 95412.248] (II) Using input driver 'mtrack' for 'Apple Inc. Magic Trackpad 2'
[ 95412.248] (**) Apple Inc. Magic Trackpad 2: always reports core events
[ 95412.248] (**) Apple Inc. Magic Trackpad 2: always reports core events
[ 95412.248] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:05AC:0265.0013/input/input23/mouse1"[ 95412.249] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 13)[ 95412.249] (II) device control: init
[ 95412.249] (**) Option "Device" "/dev/input/mouse1"
[ 95412.249] (EE) mtrack: cannot configure device
[ 95412.249] (EE) Couldn't init device "Apple Inc. Magic Trackpad 2"
[ 95412.249] (II) UnloadModule: "mtrack"
```

p2rkw, 2018-03-11:
Some reaserch done.

Hardware drivers are not in mainline kernel yet, you have to compile module and have it loaded before Xorg start:
https://github.com/torvalds/linux/pull/332

I guess with hid_magictrackpad2 mtrack will operate with event*.

majuscule, 2018-03-11:
Thank you for the research. I have compiled and installed the hid_magictrackpad2 driver:

```
❯ pacman -Ql hid-magictrackpad2-dkms
hid-magictrackpad2-dkms /usr/
hid-magictrackpad2-dkms /usr/src/
hid-magictrackpad2-dkms /usr/src/hid-magictrackpad2-4.10+hid-magictrackpad2/
hid-magictrackpad2-dkms /usr/src/hid-magictrackpad2-4.10+hid-magictrackpad2/Makefile
hid-magictrackpad2-dkms /usr/src/hid-magictrackpad2-4.10+hid-magictrackpad2/dkms.conf
hid-magictrackpad2-dkms /usr/src/hid-magictrackpad2-4.10+hid-magictrackpad2/hid-magictrackpad2.c
```

There doesn't appear to be any change, however there is currently only bluetooth support so that's likely why. Unfortunately I have other bluetooth problems at the moment... so I will  try to sort that out and come back to update here afterwards. I am wondering though - is there any way that you know of to ensure that the new driver is being loaded at all? And I'm assuming that the hid_magictrackpad2 driver is a kernal driver, whereas mtrack is an xorg driver, which explains why the former does not appear in the Xorg logs.

p2rkw, 2018-03-11:
Try
lsmod | grep magictrackpad

closed 2018-12-06 by p2rkw