p2rkw/xf86-input-mtrack #64: Pointer resets to top left corner

vilius-v, 2018-03-09:
I am having an issue where my pointer resets to the top left corner (0,0) every time I try to scroll. I will be scrolling around and if I lift my finger the pointer stays at its current location, but the next input will start it from (0,0) again. 

Has anyone seen this behavior before? I am on a FocalTech touchpad.

p2rkw, 2018-03-10:
Please set both ScrollCoastDuration and ScrollCoastEnableSpeed to 0.
I'm curious is it related to coasting, thanks.

vilius-v, 2018-03-14:
That did not work. Also, I may have worded this confusingly. It happens anytime I try to move the pointer, not just during scrolling. Sorry about that.

p2rkw, 2018-03-14:
Do you have focaltech dkms[1] installed?

[1]https://askubuntu.com/questions/609892/touchpad-not-recognized-on-asus-n550jk-tp500l-focaltech

vilius-v, 2018-03-14:
Focaltech support is in the mainline kernel since 4.01: (https://wiki.archlinux.org/index.php/ASUS_Zenbook_UX303). 

Also, I am on Arch with kernel 4.15.8 on KDE Plasma 5.12.3. The touchpad works properly with both libinput and synaptics drivers.

p2rkw, 2018-03-14:
Ok, please post Xorg.log.

vilius-v, 2018-03-15:
Here you go: [Xorg.0.log](https://github.com/p2rkw/xf86-input-mtrack/files/1813744/Xorg.0.log)

p2rkw, 2018-03-15:
I will try to disable/ignore some capabilities of my trackpad and reproduce (later).

p2rkw, 2018-03-15:
I tried to, but even with restricted caps I'm unable to reproduce.

Did you tried with unplugged digitizer and other 'pointer' devices?

vilius-v, 2018-03-16:
Digitizer is the touchscreen, not really willing to take apart my laptop to disable it, unless you know a way to disable it through software? The other pointer is a Logitech M510. Still the same issue with it removed...

vilius-v, 2018-03-16:
Figured it out. Forgot I had the libinput-gestures package. Was probably conflicting. Also using the new zenbook example you uploaded.

closed 2018-03-16 by vilius-v