p2rkw/xf86-input-mtrack #65: Multi-finger click are treated as button1-clicks

WhyNotHugo, 2018-03-10:
When two-finger clicking, or three-finger clicking, I get mouse1-clicks.

Ironically, if I rest my entire palm on the trackpad, and then click, I get a proper right-click. I'm on a 2017 MBP.

```
$ xinput --list-props 12
Device 'Apple SPI Touchpad':
        Device Enabled (136):   1
        Coordinate Transformation Matrix (138): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Tapping Enabled (273): 0
        libinput Tapping Enabled Default (274): 0
        libinput Tapping Drag Enabled (275):    1
        libinput Tapping Drag Enabled Default (276):    1
        libinput Tapping Drag Lock Enabled (277):       0
        libinput Tapping Drag Lock Enabled Default (278):       0
        libinput Tapping Button Mapping Enabled (279):  1, 0
        libinput Tapping Button Mapping Default (280):  1, 0
        libinput Natural Scrolling Enabled (281):       1
        libinput Natural Scrolling Enabled Default (282):       0
        libinput Left Handed Enabled (283):     0
        libinput Left Handed Enabled Default (284):     0
        libinput Accel Speed (285):     0.000000
        libinput Accel Speed Default (286):     0.000000
        libinput Scroll Methods Available (287):        1, 1, 0
        libinput Scroll Method Enabled (288):   1, 0, 0
        libinput Scroll Method Enabled Default (289):   1, 0, 0
        libinput Click Methods Available (290): 1, 1
        libinput Click Method Enabled (291):    1, 0
        libinput Click Method Enabled Default (292):    1, 0
        libinput Middle Emulation Enabled (293):        0
        libinput Middle Emulation Enabled Default (294):        0
        libinput Send Events Modes Available (258):     1, 1
        libinput Send Events Mode Enabled (259):        0, 0
        libinput Send Events Mode Enabled Default (260):        0, 0
        libinput Disable While Typing Enabled (295):    1
        libinput Disable While Typing Enabled Default (296):    1
        Device Node (261):      "/dev/input/event6"
        Device Product ID (262):        0, 0
        libinput Drag Lock Buttons (297):       <no items>
        libinput Horizontal Scroll Enabled (298):       1
```

```
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier "Touchpads"
    Driver "mtrack"

    Option "IgnoreThumb" "yes"
    Option "IgnorePalm"  "yes"

    # Too many false positives. :(
    Option "TapButton1" "0"

    Option "SwipeDistance" "300"
    Option "SwipeUpButton" "16"
    Option "SwipeRightButton" "17"
    Option "SwipeDownButton" "18"
    Option "SwipeLeftButton" "19"

    # Natural scrolling
    Option "ScrollUpButton"   "5"
    Option "ScrollRightButton" "6"
    Option "ScrollDownButton" "4"
    Option "ScrollLeftButton" "7"
EndSection
```

Any hints on what might be wrong here?

p2rkw, 2018-03-10:
You are using libinput.

closed 2018-03-10 by p2rkw

WhyNotHugo, 2018-03-10:
Any hints of what I ought to do for the driver to pick up, rather than libinput? I basically copied my entire config from a 2015 MBP where mtrack was working fine.

p2rkw, 2018-03-10:
https://github.com/p2rkw/xf86-input-mtrack/issues/21#issuecomment-272891450

WhyNotHugo, 2018-03-15:
Thanks! This worked perfect!