p2rkw/xf86-input-mtrack #66: Cursor "trembles

WhyNotHugo, 2018-03-16:


WhyNotHugo, 2018-03-17:
Example: https://imgur.com/a/np1Z2

Sometimes is it trembles more, sometimes less.

p2rkw, 2018-03-17:
Build this https://github.com/p2rkw/xf86-input-mtrack/tree/filter_fuzz
and fine tune SN_COORD: https://github.com/p2rkw/xf86-input-mtrack/blob/filter_fuzz/src/capabilities.c#L29

WhyNotHugo, 2018-04-02:
Sorry for the delay replying. I was out for a couple of week and couldn't get around to trying this before.

I tried that branch and the issue seems to be gone. I don't see any side effects either, so didn't need to fine-tune the value you mentioned any further.

p2rkw, 2018-04-02:
There's still problem with fuzzing, because with this path I can't draw straight, long line at 45 degrees: line have step-like patterns. Please check it out in some drawing application.

WhyNotHugo, 2018-04-02:
It looks pretty much okay to me (there's an obvious lack of anti-aliasing, but it kinda looks as expected):

[image]

p2rkw, 2018-04-03:
Your lines are wave'y just like mine.

Probably all 3 SN_* values should be increased.