p2rkw/xf86-input-mtrack #67: Integrated Button Right Click

vilius-v, 2018-03-17:
My integrated buttons don't work correctly. The right button does not perform a right click. They both perform left clicks. Any ideas?

I have the bottom edge set to zero.

xavierhd, 2018-03-29:
By integrated button, I assume you mean clickpad, in which case I think the feature is not yet implemented. But for now you could use the `ButtonZonesEnable` option in you config. It is not ideal since it use the full height of the pad.  

I was thinking about adding the option, but time as always is missing, maybe @p2rkw got a clue on how to fix it quick.

p2rkw, 2018-03-30:
That sounds like regression or incomplete implementation.

@xavierhd:
From this perspective it looks like execution always enter button0_touch>0 branch (because lowest and latest == 0).
Maybe instead of ClickFinger0, we should rename it to IntegratedButton0, and and new options for more   physical buttons (IntegratedButton1, IntegratedButton2).

What do you think?

xavierhd, 2018-04-02:
Renaming would be a good idea, but it is not a regression. OP said : 

> I have the bottom edge set to zero.

Which would not work with `ClickFinger0`.  

I think OP is referring to a physical click on the trackpad's zone he expect a left click to be produced. It is not yet implemented.
As of right now, the only way to produce a left click is by using `ButtonZonesEnable` OR say `ClickFinger2` binded to left click.

xavierhd, 2018-04-02:
@vilius-v, the right button you refer to, does you mean:

- mean bottom right part of your touch pad [image]
- or a separate button
[image]

vilius-v, 2018-04-02:
The bottom right part, as in the first image.

xavierhd, 2018-05-11:
Just so you know, i'm working on something.
I'm planning to use the "zone" feature to split the pad.
To do: 
 * Limit the zone height.
   - Use the `EdgeBottom` config.
   - or new config parameter? I think i'll go with that.
 * Add a configuration for the number of `zones` since if you set all 3 `ClickFinger` param inside the config, you are forced to use 3 zones.

xavierhd, 2018-06-06:
Hey, here is the update!

As stated in the previous post, I made it possible to set a zone in the bottom which is binded to it's own click event.
----
Introducing a couple of new params (and their default value):
 - `Option "ZoneButton1" "1"`
Button to use in zone 1
 - `Option "ZoneButton2" "3"`
Button to use in zone 2
 - `Option "ZoneButton3" "0"`
Button to use in zone 3
 - `Option "ButtonZonesInEdgeBottom" "true"`
Restrict button zones in the bottom edge. You need the option `EdgeBottomSize` to be bigger than 0. Read EdgeBottomSize doc fo more detail.
---
Note that zones behavior change, with this update, the number of zones depend of the number of `ZoneButton#` you set to more than 0.

p2rkw, 2018-06-23:
@vilius-v : Does it work for you on **master** branch with following config:
```
        Option "EdgeBottomSize"	"5"  # dont count touches from bottom 5% of area
        Option "ButtonEnable" "true"
        Option "ButtonIntegrated" "true"
        Option "ButtonZonesEnable" "true"
        Option "ClickFinger1" "1"
        Option "ClickFinger2" "2"
        Option "ClickFinger3" "3"
        Option "FingerLow" "0"
        Option "FingerHigh" "0"
```
?

rodrigogolive, 2018-11-10:
hi, news about this issue? I'm new on mtrack and the last thing that is missing for me is to bind the right-click on the right-most corner of touchpad (the "ButtonZonesInEdgeBottom" seems to be a fix for the issue for me)

or, adding my 2 cents, just one option to respect the height of the EdgeBottonSize on combination with ButtonZonesEnable could be great

thanks

xavierhd, 2018-11-11:
Hi Rodrigo!
You could checkout e5fe2fdf9556b37766ba3c2dd6f595705412ced7 which is the pull request made to add the feature.  
See https://github.com/xavierhd/xf86-input-mtrack/tree/zone_high#enabling-soft-button for more configuration hints about the new feature.

p2rkw doesn't seem to be very responsive since last summer.

p2rkw, 2018-11-11:
Hi @xavierhd : sorry I forgot to publish my comments in your PR.

 I needed some time to understand new code.

rodrigogolive, 2018-11-12:
cool, will tests it asap ;)

rodrigogolive, 2018-11-12:
> Hi Rodrigo!
> You could checkout [e5fe2fd](https://github.com/p2rkw/xf86-input-mtrack/commit/e5fe2fdf9556b37766ba3c2dd6f595705412ced7) which is the pull request made to add the feature.
> See https://github.com/xavierhd/xf86-input-mtrack/tree/zone_high#enabling-soft-button for more configuration hints about the new feature.
> See the last line : Option "ButtonZonesInEdgeBottom" "true"

cool, will test it asap ;)

HactarCE, 2019-01-17:
Is it possible to make a "hard" click outside of the bottom zone be interpreted as left click, without sacrificing the left/middle/right click within the bottom zone? i.e.
[image]

closed 2019-03-16 by p2rkw