p2rkw/xf86-input-mtrack #73: continue action after reaching touchpad edge?

roufamatic, 2018-06-18:
Apologies if this isn't the best place to ask about this.

I am a big fan of 3-finger drag and it was the hardest thing to let go of when moving off a Mac to a Linux laptop. This project is awesome, and following [this guide](http://martys.typepad.com/blog/2015/08/3-finger-drag-on-linux.html) I was able to get it up and running quickly. 

The one thing I haven't figured out how to replicate is the ability to continue the action after reaching the edge of the touchpad. The OS X implementation doesn't release the drag action until a certain amount of time has passed with no additional touch input. This allows you to reach the edge of the touchpad, lift up your fingers, place them back down on the other side, and continue dragging. (A simple use case for this is trying to transition from Google Maps to Street View, where you have to drag pegman from the bottom right corner to some street on the top left -- there just isn't enough trackpad to get there.)

Is there a setting that I can use to get this behavior out of xf86-input-mtrack? Or is TapDrag the only option?

xavierhd, 2018-06-19:
Are you looking for *TapDragLockTimeout* option?

p2rkw, 2018-06-19:
I was about to write similar reply, but after few glances at the code I think it will work only with tap-do-drag drag. Worth to check anyways :)

xavierhd, 2018-06-21:
Or else you could also use the `Hold1Move1Stationary` option which let you keep a finger on the pad and move another finger to drag. The drag state won't be released until you move/remove the stationary finger.

But yet, the behavior from OSX would be neat.
While in drag state,
If the finger is lifted on the edge,
Lock drag state for X seconds.

Maybe one day...

joeyave, 2020-11-10:
How to make pointer moving when it reaches the edge like it's done on Windows?