p2rkw/xf86-input-mtrack #76: dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2

toszter, 2018-09-05:
I'm just trying an mtrack installation because the libinput with dkms was a total flop (I couldn't move, only click and drag).

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.1 LTS
Release:        18.04
Codename:       bionic

$ uname -r
4.18.5-041805-generic

$ dkg-buildpackage
dpkg-buildpackage: info: source package xserver-xorg-input-mtrack
dpkg-buildpackage: info: source version 0.5.0
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Paweł Turkowski <p2rkw0@gmail.com>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build xf86-input-mtrack
 fakeroot debian/rules clean
dh_testdir
dh_testroot
/usr/bin/make clean
make[1]: Entering directory '/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack'
test -z "mtrack_drv.la" || rm -f mtrack_drv.la
rm -f ./so_locations
rm -rf .libs _libs
rm -rf driver/.libs driver/_libs
rm -rf src/.libs src/_libs
 rm -f mtrack-test
rm -f *.o
rm -f driver/*.o
rm -f driver/*.lo
rm -f src/*.o
rm -f src/*.lo
rm -f tools/*.o
rm -f *.lo
make[1]: Leaving directory '/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack'
dh_clean
dh_clean: Compatibility levels before 9 are deprecated (level 5 in use)
 dpkg-source -b xf86-input-mtrack
dpkg-source: warning: no source format specified in debian/source/format, see dpkg-source(1)
dpkg-source: info: using source format '1.0'
dpkg-source: warning: source directory 'xf86-input-mtrack' is not <sourcepackage>-<upstreamversion> 'xserver-xorg-input-mtrack-0.5.0'
dpkg-source: info: building xserver-xorg-input-mtrack in xserver-xorg-input-mtrack_0.5.0.tar.gz
dpkg-source: info: building xserver-xorg-input-mtrack in xserver-xorg-input-mtrack_0.5.0.dsc
 debian/rules build
dh_testdir
/usr/bin/make
make[1]: Entering directory '/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack'
/usr/bin/make  all-am
make[2]: Entering directory '/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack'
  CC       src/capabilities.lo
  CC       src/gestures.lo
  CC       src/hwstate.lo
  CC       src/mconfig.lo
  CC       src/mtouch.lo
  CC       src/mtstate.lo
  CC       src/trig.lo
  CC       driver/mtrack.lo
  CC       driver/mprops.lo
  CCLD     mtrack_drv.la
  CC       src/mtrack_test-capabilities.o
  CC       src/mtrack_test-gestures.o
  CC       src/mtrack_test-hwstate.o
  CC       src/mtrack_test-mconfig.o
  CC       src/mtrack_test-mtouch.o
  CC       src/mtrack_test-mtstate.o
  CC       src/mtrack_test-trig.o
  CC       tools/mtrack_test-mtrack-test.o
  CCLD     mtrack-test
src/mtrack_test-hwstate.o: In function `hwstate_modify':
/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack/src/hwstate.c:126: undefined reference to `mtdev_get'
src/mtrack_test-mtouch.o: In function `mtouch_open':
/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack/src/mtouch.c:42: undefined reference to `mtdev_open'
src/mtrack_test-mtouch.o: In function `mtouch_close':
/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack/src/mtouch.c:70: undefined reference to `mtdev_close'
collect2: error: ld returned 1 exit status
Makefile:565: recipe for target 'mtrack-test' failed
make[2]: *** [mtrack-test] Error 1
make[2]: Leaving directory '/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack'
Makefile:419: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/azatar/Engineering/ubuntu/touchpad/xf86-input-mtrack'
debian/rules:8: recipe for target 'build-arch' failed
make: *** [build-arch] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2

What's the problem, I wonder?

p2rkw, 2018-11-04:
Please try sudo apt build-dep xserver-xorg-input-mtrack beforehand.
Probably libmtdev1 package is missing.

closed 2018-12-06 by p2rkw

vcavallo, 2020-03-23:
I'm having the same issue. Running `build-dep` returns:

```
$ sudo apt build-dep xserver-xorg-input-mtrack
Reading package lists... Done
E: You must put some 'source' URIs in your sources.list
```

I tried installing `libmtdev1` directly and running `make` again. no luck.

Then I installed `libmtdev-dev` directly and ran `make` again and still no luck - _but_ I re-ran the `./configure --with-xorg-module-dir=/usr/lib/xorg/modules` command and tried `make` again and it succeeded (as well as `make install`, subsequently.)