p2rkw/xf86-input-mtrack #77: improve 3-finger-drag to 3-finger-initiated-drag

complyue, 2018-09-06:
As many others I found mtrack is the only solution to enable 3-finger-drag for non-osx laptops. And I missed it badly when started using my matebook pro x with Ubuntu 16.04.

But after used current solution for a while, I discovered that it's very easy to disturb the dragging procedure, coz my 3 fingers tend to become closer after some distance, and the clickpad driver may flash between 2-points touch and 3-points touch or even single-point touch for its report. This will break the  intuition of the drag at all, as the result is like cancelling & restart dragging at some middle path points. The workaround is to spread my 3 fingers wide enough during the entire dragging course.

But I'm never enforced to spread my fingers during 3-finger drag at my good old macbook pro, per my verification with my magic trackpad attached to my old iMac (my mbp had dead and I primarily work on my mac pro with a mighty mouse, since two 4k display plus one 1080p display is used, trackpad is never suitable for routinely long travel of mouse cursors), what OSX does is what I would name it `3-finger-initiated-drag` instead of `drag-by-3-finger`, i.e. after the drag has started by 3-finger, I can release any other one or two fingers and still continue the drag gesture, this is quite ideal than current mtrack's implementation.

And I modified the code for my personal use with `3-finger-initiated-drag` implemented at [this branch](https://github.com/complyue/xf86-input-mtrack/tree/optmz-drag3):
change set: https://github.com/complyue/xf86-input-mtrack/commit/c343d0671fc22b5e280e5de2829f99b8dce1129c

it also incorporated a new option "MinTapTime" as I found my self quite often touching the clickpad surface by accident when start typing, most of the touches are very short so a min contact time can filter them out.

But I don't think my mod in its current shape is suitable for a PR, as unfortunately I have to disable rotate/scale altogether to achieve my goal, although I don't find rotate/scale useful on Ubuntu desktops, other may still need then.

I hope people here may review my changes to figure out whether it worth to go into the official release.

ps my config: https://github.com/complyue/xf86-input-mtrack/blob/optmz-drag3/examples/matebook-pro-x.conf
```
# https://github.com/p2rkw/xf86-input-mtrack
# https://github.com/complyue/xf86-input-mtrack/tree/optmz-drag3
Section "InputClass"

    Identifier      "Touchpads w/ mtrack"

    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "mtrack"

EndSection

Section "InputClass"

    Identifier "mtrack: Compl's tweaks"

    MatchIsTouchpad "on"

    MatchDriver "mtrack"

    # Disable tap-hold-move drag
    Option "TapDragEnable" "false"
    Option "TapDragTime" "0"

    # Enable 3-finger drag
    Option "SwipeDistance" "1"
    Option "SwipeLeftButton" "1"
    Option "SwipeRightButton" "1"
    Option "SwipeUpButton" "1"
    Option "SwipeDownButton" "1"
    Option "SwipeClickTime" "0"
    Option "SwipeSensitivity" "1000"

    # Enable 4-finger swipe
    # Extra: 
    #   enable workspaces in "System Settings" -> "Appearance"
    #   apt install compizconfig-settings-manager then import MissionCtrl.profile
    #   make sure "Desktop" -> "Desktop Wall" and Expo is checked
    Option "Swipe4Distance" "60"
    Option "Swipe4ClickTime" "500"
    Option "Swipe4Sensitivity" "0"  # no cursor move
    Option "Swipe4UpButton" "8"
    Option "Swipe4DownButton" "9"
    Option "Swipe4LeftButton" "10"
    Option "Swipe4RightButton" "11"

    # Making tap (especially 3-finger) click easy
    Option "ButtonTouchExpire" "160"
    Option "MaxTapTime" "160"
    # Making accelerated move/drag easy, default value is too big such that short touch does not initiate acceleration at all
    Option "MaxTapMove" "30"
    # Compl's patch to filter out touch by accident
    Option "MinTapTime" "30"

    # Better acceleration feels
    # https://www.x.org/wiki/Development/Documentation/PointerAcceleration/
    Option "AdaptiveDeceleration" "2"
    Option "AccelerationProfile" "2"

    # Sensitivity controls how fast your cursor will move. 1 is the default
    Option          "Sensitivity" "1"
    # This ignores tap-to-click, which causes more problems than benefit in my experience
    Option          "TapButton1" "1"
    Option          "TapButton2" "3"
    Option          "TapButton3" "2"
    # If you want a middle-click, then "ClickFinger2" should be value "2"
    Option          "ClickFinger1" "1"
    Option          "ClickFinger2" "3"
    Option          "ClickFinger3" "2"
    Option          "ButtonMoveEmulate" "true"
    Option          "ButtonIntegrated" "true"
    Option      "ButtonEnable" "true"
    # No button emulation by zones
    Option          "ButtonZonesEnable" "false"
    Option          "ClickTime" "25"
    # Ensures that bottom 5% of touchpad doesn't register taps
    Option          "EdgeBottomSize" "5"
    # ScrollCoast makes touchpad feel a bit more Mac-like, although it coasts in chunks and isn't relative to speed at which two finger scroll was happening
    Option          "ScrollSmooth" "1"
    Option          "ScrollCoastDuration" "800.0"
    Option          "ScrollCoastEnableSpeed" "0.2"
    # This sets up Macbook-like natural scroll. If you want to scroll down by swiping your fingers down, reverse the "5" and the "4" here:
    Option          "ScrollUpButton" "5"
    Option          "ScrollDownButton" "4"
    Option          "ScrollLeftButton" "7"
    Option          "ScrollRightButton" "6"
    # Without this option set to a high value, there are types of click+hold-and-move functionality (most easily reproed by click and then move up-right) that get ignored
    Option          "Hold1Move1StationaryMaxMove" "1000"
    # Smaller ScrollDistance translates to faster scrolling. ScrollDistance of 10 scrolls a long page in one swipe.
    Option          "ScrollDistance" "22"
    Option      "ScrollClickTime" "12"
    Option      "ScrollSensitivity" "0"

EndSection
```

p2rkw, 2018-11-04:
Hi,
thanks for sharing, 
accidentally I already got '3-finger-initiated-drag' implemented in my working tree (but not pushed yet).
Probably I started tests and forgot to push. 

Thanks anyway.

complyue, 2018-11-05:
when will you release your implementation? I found my mod still got unexpected cursor jumps when my 3 fingers is some close to each others, hope your official implementation can address such issues.

closed 2018-11-05 by complyue

gch1p, 2020-12-04:
I double the question:)

@p2rkw can you please push your implementation somewhere? If it's not polished enough yet and you have no time for that, community will eventually make it good enough for merging, I think.

complyue, 2020-12-05:
FYI, I ended up resort to libinput, which overcomes many glitches of the hardware so feels way much more smoother overall. The 3 finger feature is neither official there, but customized branches work well enough. See https://gitlab.freedesktop.org/libinput/libinput/-/issues/298#note_708088