p2rkw/xf86-input-mtrack #93: Move first finger, without releasing the first finger press the second and get left mouse click

voidEvgeny, 2019-08-28:
This works on macos and windows but not in Linux...
Move first finger, without releasing the first finger press the second and get a button 1 pressed (left mouse button click), can this be configured?

p2rkw, 2019-09-07:
https://github.com/p2rkw/xf86-input-mtrack#Hold1Move1StationaryButton

voidEvgeny, 2019-09-07:
> https://github.com/p2rkw/xf86-input-mtrack#Hold1Move1StationaryButton

I do not need to select the text, I need to move the cursor with one finger without releasing this finger, tap the second finger and get button1 pressed. Please add such functionality to your driver, this is the best touchpad driver that I used.

a11ce, 2022-11-16:
Seconding this. It works with clicking, but not tapping.