p2rkw/xf86-input-mtrack #94: Help with scroll coasting not working

djhoese, 2019-09-06:
I just built and install mtrack from this master branch. I was able to get almost every basic use I wanted to work except scroll coasting. Scrolling was originally very slow but reduced the scroll distance and that seemed to help. I've tried a few different things, but I can't seem to get the scroll to continue no matter how fast I do the scroll or the distance that I do it. I tried looking at the source code, but got lost since it looks like some of the options are hardcoded and the configured ones seem (?) to be ok.

Anyone have ideas?

```
$ xinput list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ SYNA1202:00 06CB:CD65 Touchpad            id=12   [slave  pointer  (2)]
⎜   ↳ PS/2 Synaptics TouchPad                   id=15   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Power Button                              id=9    [slave  keyboard (3)]
    ↳ Sleep Button                              id=10   [slave  keyboard (3)]
    ↳ BisonCam,NB Pro: BisonCam,NB Pr           id=11   [slave  keyboard (3)]
    ↳ Intel HID events                          id=13   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=14   [slave  keyboard (3)]
```

```
$ xinput list-props 12
Device 'SYNA1202:00 06CB:CD65 Touchpad':
        Device Enabled (184):   1
        Coordinate Transformation Matrix (186): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (318):     0
        Device Accel Constant Deceleration (319):       1.000000
        Device Accel Adaptive Deceleration (320):       1.000000
        Device Accel Velocity Scaling (321):    10.000000
        Trackpad Disable Input (322):   0
        Trackpad Sensitivity (323):     1.000000
        Trackpad Touch Pressure (324):  5, 1
        Trackpad Button Settings (325): 1, 0
        Trackpad Button Emulation Settings (326):       0, 1, 100, 0
        Trackpad Button Emulation Values (327): 0, 1, 1, 1, 1, 2, 0
        Trackpad Tap Settings (328):    25, 120, 400
        Trackpad Tap Button Emulation (329):    1, 3, 2, 0
        Trackpad Thumb Detection (330): 0, 0
        Trackpad Thumb Size (331):      25, 70
        Trackpad Palm Detection (332):  1, 0
        Trackpad Palm Size (333):       40
        Trackpad Gesture Settings (334):        10, 100
        Trackpad Smooth Scroll (335):   1
        Trackpad Scroll Settings (336): 50, 20, 0
        Trackpad Scroll Buttons (337):  5, 4, 7, 6
        Trackpad Swipe Settings (338):  1, 0, 1000
        Trackpad Swipe Buttons (339):   1, 1, 1, 1
        Trackpad Swipe4 Settings (340): 700, 300, 0
        Trackpad Swipe4 Buttons (341):  0, 0, 0, 0
        Trackpad Scroll Coasting (342): 0.100000, 500.000000
        Trackpad Edge Scroll Settings (343):    105, 20, 0, 4, 5, 6, 7
        Trackpad Edge Sizes (344):      0, 10, 0, 0
        Trackpad Scale Distance (345):  150
        Trackpad Scale Buttons (346):   12, 13
        Trackpad Rotate Distance (347): 150
        Trackpad Rotate Buttons (348):  14, 15
        Trackpad Hold1Move1 Stationary Settings (349):  20, 1
        Trackpad Hold1Move1 Settings (350):     1, 0, 1000
        Trackpad Hold1Move1 Buttons (351):      1, 1, 1, 1
        Trackpad Drag Settings (352):   1, 350, 40, 200, 500
        Trackpad Axis Inversion (353):  0, 0
```

nickberry17, 2019-11-10:
Did you set the ScrollCoastDuration to a high enough value?  Try setting your ScrollCoastEnableSpeed to something like .3

```
        Option          "ScrollCoastDuration" "1000"
        Option          "ScrollCoastEnableSpeed" ".3"
```

djhoese, 2019-11-10:
I had it set to 0.3 for EnableSpeed and the duration to 500. I bumped it to 1000 and yeah it feels a little better.

nickberry17, 2019-11-10:
You can try bumping ScrollCoastDuration, I believe values over 1s will give you that coasting feeling.  What we need now is a way to more gradually ramp up and down the coasting speed, i.e. overcome initial friction (a _weight_ value?), inertia coasts you, then friction slows you back down.  
For now you can try my Macbook Pro 2013 config. https://github.com/p2rkw/xf86-input-mtrack/pull/95

nickberry17, 2019-11-11:
I was completely wrong I think.  Turns out for me I had libinput also running.  I moved the mtrack.conf file to a higher priority than libinput and now things are much different (better).

Try something like this on your system:
`mv /usr/share/X11/xorg.conf.d/50-mtrack.conf /usr/share/X11/xorg.conf.d/30-mtrack.conf`