p2rkw/xf86-input-mtrack #99: Scrolling during scroll coast results in very fast scrolling

kokowang, 2020-01-23:
Scrolling again while coasting results in scroll speed becoming extremely fast temporarily. Same issue as https://github.com/BlueDragonX/xf86-input-mtrack/issues/125

This tends to not happen with the default scroll coast duration, setting the scroll coast duration to 1000ms helps makes the problem easier to demonstrate.

Built on 1/22/2020, Macbook Pro 2019 (15,1) | Arch Linux | KDE
Config File:
```Section "InputClass"
MatchIsTouchpad "on"
Identifier      "Touchpads"
MatchDevicePath "/dev/input/event*"
Driver          "mtrack"
# Sensitivty and thresholds
Option          "Sensitivity" "0.2"
Option          "FingerHigh" "8"
Option          "FingerLow" "8"
Option          "IgnoreThumb" "false"
Option          "ThumbRatio" "70"
Option          "ThumbSize" "10"
Option          "IgnorePalm" "true"
Option          "PalmSize" "32"
# Disable annoying taps
Option          "TapButton1" "0"
Option          "TapButton2" "0"
Option          "TapButton3" "0"
Option          "TapDragEnable" "false"
# Clicks
Option          "ClickFinger1" "1"
Option          "ClickFinger2" "3"
Option          "ClickFinger3" "2"
Option          "ButtonMoveEmulate" "true"
Option          "ButtonIntegrated" "true"
Option          "ButtonEnable" "true"
Option          "ClickTime" "25"
# Swiping
Option          "SwipeLeftButton" "8"
Option          "SwipeRightButton" "9"
Option          "SwipeUpButton" "0"
Option          "SwipeDownButton" "0"
Option          "SwipeDistance" "700"

Option          "ScrollUpButton" "5"
Option          "ScrollDownButton" "4"
Option          "ScrollLeftButton" "7"
Option          "ScrollRightButton" "6"
# Just.... don't. Please
Option          "Hold1Move1StationaryButton" "0"
Option          "EdgeScrollLeftButton" "0"
Option          "EdgeScrollRightButton" "0"
Option          "EdgeScrollUpButton" "0"
Option          "EdgeScrollDownButton" "0"
# Scrolling
Option          "ScrollDistance" "150"
Option          "ScrollClickTime" "0"
Option          "ScrollSensitivity" "0"
Option          "ScrollCoastDuration" "1000"
Option          "ScrollCoastEnableSpeed" "0.02"

EndSection
```

gipert, 2020-02-11:
Same issue on my MacBookAir 2012 / Arch Linux

retesta, 2020-05-08:
Same issue on LG Gram 2018 / Ubuntu 18.04 LTS

hisbaan, 2020-06-02:
Same here on a Dell XPS 13 9300 / Arch Linux & MacBook Air 2015 / Arch Linux

zsdiogo, 2020-12-04:
Same on a MacbookPro 2015 running Ubuntu 20.10.

mdlockyer, 2021-03-01:
Same here for my MacBook Pro. If you start a new 2 finger scroll during the coast, you get blasted half way across the page.

artefom, 2021-03-10:
Can indeed confirm that this is the case. Looks like a bug. Possibly coasting timer does not get disabled after tapping again during coasting, so scrolling gets applied twice?

artefom, 2021-03-10:
Will look into the issue, try to debug. If anyone have any ideas pls share

p2rkw, 2021-03-25:
Hi, you may start here: https://github.com/p2rkw/xf86-input-mtrack/blob/master/src/gestures.c#L720

hxss, 2021-11-06:
Same issue on HP 435 g8. Intercepts/joggle per scrolling becomes impossible