p2rkw/xf86-input-mtrack #107: Two finger scroll direction

wolkym, 2020-06-04:
Hello guys, thanks for the driver! It helps a lot with Apple Trackpad 2 on 4.18 kernel.
I have troubles to switch scroll direction, move two fingers down and scroll down at the same time.
There are buttons for scroll up and down (default 4 and 5).

```
ScrollUpButton For two finger scrolling. The button that is triggered by scrolling up.
Integer value. A value of 0 disables scrolling up. Defaults to 4.

ScrollDownButton For two finger scrolling. The button that is triggered by scrolling down.
Integer value. A value of 0 disables scrolling down. Defaults to 5.
```

I tried to swap it, but it did not help.
Is there any way to change direction of two-finger scroll?

Dimich-x33, 2020-06-18:
Same issue.

p2rkw, 2020-11-15:
Hi, I thought I fixed that, please set ScrollSmooth to 0 and then experiment with swapping ScrollUpButton and ScrollDownButton one more time.